TixiaoShan/LIO-SAM #220: "Large velocity, reset IMU-preintegration!"

heirenlop, 2021-03-23:
Hi @TixiaoShan,
Thanks for sharing your excellent work.
1. I am testing the lio-sam system but I got a warning for "Large velocity, reset IMU-preintegration!". And the trajectory result is very bad.

[image]

2. My IMU data is from INS570D and lidar data is from RS-LIDAR 16, the default coordinate of IMU is "x-forward, y-right, z-down" and the default coordinate of RS-LIDAR is "x-right, y-forward, z-up", i have changed both of them to "x-forward, y-left, z-up", but the result is still bad.

[image]
[image]

As shown in the two figures, I modified the code for IMU and LIDAR to complete coordinate transformation.

3.I used lidar_imu_calib to calib LIDAR and IMU. I did it in a small room, but when I put the extrinsicRot and extrinsicRPY into the yaml, the result did not change, still very bad. 

[image]

4. I don't know which part I did wrong. I hope someone can give me some advice，such as whether I need to calibrate IMU with IMU_utils?

closed 2021-03-25 by heirenlop

duanchengwen, 2021-03-31:
@heirenlop I ran into the same problem, did you solve it

ibrahimhroob, 2021-04-16:
@duanchengwen @heirenlop Hi guys, I am getting the same issue, any luck so far? Thanks

bobusfil, 2021-04-30:
Same issue

bobusfil, 2021-04-30:
I solved it by following the instructions in the Readme.md ... where they instruct to try different "extrinsicRot" and extrinsicRPY in "params.yaml." There are two versions and one of them is commented out. So try uncomment the commented one and comment the ... not commented one. .. 
Hope it helps.

ibrahimhroob, 2021-05-02:
Thanks for the suggestion. I have tried that, but I think my issue with the imu plugin in gazebo since I am using a simulated environment to test the algorithm. For some reason the imu plugin is very noisy and the output data are all over the place even when the robot is stand still! I have set the Gaussian noise to 0, but that did not change anything! 

This is the plugin configuration: 
    <gazebo reference="$(arg tf_prefix)${sensor_name}">
      <gravity>true</gravity>
      <sensor name="$(arg robot_name)imu_sensor" type="imu">
        <always_on>true</always_on>
        <update_rate>200</update_rate>
        <visualize>true</visualize>
        <topic>__default_topic__</topic>
        <plugin filename="libgazebo_ros_imu_sensor.so" name="(arg robot_name)imu_plugin">
          <topicName>imu_raw</topicName>
          <bodyName>${sensor_name}</bodyName>
          <updateRateHZ>200.0</updateRateHZ>
          <gaussianNoise>0.0</gaussianNoise>
          <xyzOffset>0 0 0</xyzOffset>
          <rpyOffset>0 0 0</rpyOffset>
          <frameName>$(arg tf_prefix)${sensor_name}</frameName>
          <initialOrientationAsReference>false</initialOrientationAsReference>
        </plugin>
      </sensor>
    </gazebo>

arjunskumar, 2022-02-22:
@bobusfil do you remember the output frequency of your IMU?

smileyenot983, 2022-03-21:
Hi guys, do you have any updates on this?
[image]

I also have this really strange behaviour when map is going up and down while robot is staying. I am working with turtlebot + velodyne VLP-16 simulator in gazebo simulation

ibrahimhroob, 2022-03-21:
you may need to check the imu frequency

smileyenot983, 2022-03-21:
`    <gazebo reference="imu_link">
        <gravity>true</gravity>
        <sensor name="imu_sensor" type="imu">
            <always_on>true</always_on>
            <update_rate>1000</update_rate>
            <visualize>true</visualize>
            <topic>__default_topic__</topic>
            <plugin filename="libgazebo_ros_imu_sensor.so" name="imu_plugin">
                <topicName>trunk_imu</topicName>
                <bodyName>imu_link</bodyName>
                <updateRateHZ>1000.0</updateRateHZ>
                <gaussianNoise>0.0</gaussianNoise>
                <xyzOffset>0 0 0</xyzOffset>
                <rpyOffset>0 0 0</rpyOffset>
                <frameName>imu_link</frameName>
            </plugin>
            <pose>0 0 0 0 0 0</pose>
        </sensor>
    </gazebo>`

Currently i have these settings for imu

ibrahimhroob, 2022-03-21:
I have encounter this issue before, so the was I solved it is to set the imu frequency into sensible range like 200Hz, then I increased the frequency of the simulator, I set it to 1k hz, so I would suggest to check your simulator update rate, here are my configuration as a reference: 
`
    <physics name='default_physics' default='0' type='ode'>
      <real_time_update_rate>1000</real_time_update_rate>
      <real_time_factor>1</real_time_factor>
      <max_step_size>0.001</max_step_size>
    </physics>
`
Note: the update rate frequency must be at least x2 faster then your fastest signal in the simulation. https://en.wikipedia.org/wiki/Aliasing

smileyenot983, 2022-03-21:
After i changed the extrinsics and reduced the update_rate for imu to 200 it became slightly better(it does not start oscillating or falling down from the beginning), but it is still bad. It first starts drifting a little bit, but after some time it starts oscillating a lot even though my robot stays at the same place.

[image]

ibrahimhroob, 2022-03-21:
by the way, you could try FAST-LIO: https://github.com/hku-mars/FAST_LIO it did work much better for my use case

smileyenot983, 2022-03-21:
Here is the physics of the world i am currently using:
   ` <physics name='default_physics' default='0' type='ode'>
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>1000</real_time_update_rate>
    </physics>`
    
I will definitely check fast_lio, but at the same time i'd really like to know why is this problem happening, i am also facing same drift when i am trying to run it with turtlebot

houcemsaidani, 2022-04-20:
> I have encounter this issue before, so the was I solved it is to set the imu frequency into sensible range like 200Hz, then I increased the frequency of the simulator, I set it to 1k hz, so I would [...]

Hey there Ibrahim! I'm sorry but I did my own bag file using gazebo simulateur too and when i tried to launch lio sam, the map was overlapping like the previous guy. I changed my IMU to 200hz like you said, it did not work, i tried to uncomment the "extrinsic"", lio sam launch won't work anymore and it shows this:

RLException: error loading <rosparam> tag: 
	file /home/houcem/catkin_ws/src/LIO-SAM/config/params.yaml contains invalid YAML:
while parsing a block mapping
  in "<unicode string>", line 4, column 3:
      pointCloudTopic: "points_raw"    ... 
      ^
expected <block end>, but found '<block mapping start>'
  in "<unicode string>", line 49, column 4:
       extrinsicRot: [1, 0, 0,
       ^
XML is <rosparam file="$(find lio_sam)/config/params.yaml" command="load"/>
The traceback for the exception was written to the log file

Can you help? thanks!

ibrahimhroob, 2022-04-20:
Hi @houcemsaidani , this seems to be indentation issue, so please make sure the indents are correct. If you still facing issues with lio-sam I suggest this package: https://github.com/hku-mars/FAST_LIO .. thanks

joao-aguizo, 2022-11-21:
@ibrahimhroob Were you able to deal with the flickering of your simulation's IMU in LIO-SAM?

ibrahimhroob, 2022-11-21:
@joao-aguizo yes I was long time ago, the issue was mainly with the sampling time of my simulator, so I set it to 1k Hz and set my IMU frequency to 200 Hz, with this configuration you can avoid sensor sampling aliasing.

harisbinyousaf24, 2023-06-08:
Hi @joao-aguizo Can you tell me where did you set the IMU Frequency? exactly in which script f the package?

joao-aguizo, 2023-06-08:
@harisbinyousaf24 At the time I was using the Husky model. Assuming that you are using it as well, you should take a look into [this line](https://github.com/husky/husky/blob/noetic-devel/husky_description/urdf/husky.urdf.xacro#L379), in case you wish to increase the frequency of the IMU plugin.

harisbinyousaf24, 2023-06-09:
@joao-aguizo @ibrahimhroob Actually the thing is that my data is published by ros bag i have recorded data. My imu/data is publishing at 20Hz which is i know too low frequency as per requirement. Any solution regarding this? My output is same as @smileyenot983 Would highly appreciate help at this moment!

joao-aguizo, 2023-06-11:
@harisbinyousaf24 20Hz is definitely really low for LIO-SAM... I would try to get a dataset with an higher frequency, I have worked in the past with an IMU@100Hz and it worked decently. 

If the first suggestion is not an option for you, targeting the IMU@100Hz that I mentioned before, what I have seen in the past is that you could slow down the bag and re-record all the topics except the one from the IMU, the output dataset would then be 5 times slower in every topic except the latter, which would appear to be at a rate of 100Hz relative to the others. **Be aware that with this approach you would lose part of your dataset.**

joao-aguizo, 2023-06-11:
@harisbinyousaf24 20Hz is definitely really low for LIO-SAM... I would try to get a dataset with an higher frequency, I have worked in the past with an IMU@100Hz and it worked decently. 

If the first suggestion is not an option for you, targeting the IMU@100Hz that I mentioned before, what I have seen being done in the past involves re-recording a slowed versions of all the topics except the one from the IMU, the output dataset would then be 5 times slower in every topic except the latter, which would appear to be at a rate of 100Hz relative to the others. I would start by taking a look into [ROS' topic_tools throttle](http://wiki.ros.org/topic_tools/throttle).

harisbinyousaf24, 2023-06-11:
@joao-aguizo Thank u for the suggestion! I probably would have considered that if recording data was in my hands. Actually the data that comes to me in the form of rosbag is from a different location. 
Can you please give me any other suggestion in that respect? Also if i upsample the data coming from IMU at 200Hz and then publish it on some other topic and then replace that topic in the LIO-SAM launch file would that help?

happylifelsc, 2023-11-08:
> @joao-aguizo Thank u for the suggestion! I probably would have considered that if recording data was in my hands. Actually the data that comes to me in the form of rosbag is from a different [...]

Hi have you solved the problem? I'm still not sure why the frequency of the IMU could be a problem...

Danendra10, 2025-05-24:
i've tried every single solution above, and yet I didn't find any solutions

alitekes1, 2025-09-01:
> i've tried every single solution above, and yet I didn't find any solutions

did you solve the problem? can you help me? i try to done same thing with you. You can talk about your experience even bad. I will wait your response.