TixiaoShan/LIO-SAM #223: Velodyne HDL-32E + NOVATEL SPAN CPT INS

lincob, 2021-03-27:
Hello,

I'm considering using the combination of a velodyne HDL-32E and a NOVATEL SPAN CPT INS (not the latest generation, the one with the OEM628 board).

The SPAN CPT unit itself only has a 6DOF IMU but is able to provide yaw output using the GNSS information (dynamic heading accuracy stated in datasheet : 0.035° in post processing).
The max output rate is 100Hz, I've read that it may be on the low side for LIO SAM...

I'm not interested in real time operations, I would post process GNSS/IMU data with inertial explorer before running slam.

If anybody could give me feedback about this setup it would be greatly appreciated.
Should I provide a second, faster and more accurate IMU, for slam and keep exclusively the SPAN CPT for GNSS positioning ?
How should I sync everything ? Should I sync lidar data with GPS time with the PPS signal ?
Thanks for your time.

closed 2021-06-03 by stale[bot]

MaybeShewill-CV, 2021-11-03:
@lincob Have you successed with thosed hardware you mentioned above:)

lincob, 2021-11-16:
@MaybeShewill-CV short answer : no. I ended up buying a 3dm-gx5-45 instead. The Novatel SPAN system was too demanding, it has to be fed with rtk corrections in order to output the IMU data... Since I'm not interested in real-time data and everything is post processed, it wasn't worth the pain.
I don't know what's your application but I had decent results with microstrain + hdl32, stunning visuals, but large amount of drift, especially in Z, which makes the solution very hard to rely on in terms of surveying. I've tested in very harsh environments though, since in simple cases traditional laser scanning still remains the best option.
Cheers.

jangirrishabh, 2023-10-27:
Hello @lincob I am having similar issues with 3DM-GX5-25. The problem is it does not generate a PPS signal (no GPS). But talking about your solution, if I were to go for a 3DM-GX5-45 instead, could you describe your system?

I am guessing the HDL32 is connected to the laptop through ethernet and the IMU is connected to the laptop through USB. Which device is generating the clock in this case, the IMU or the laptop? Any description would be greatly appreciated.

lincob, 2024-09-14:
@jangirrishabh sorry, totally missed this one... Correct, I'm using a raspberry pi, HDL32 ethernet, GX5-45 usb. The PPS output of the GX5 to the PPS input of the HDL32 and I'm sending a NMEA message at 1Hz with a custom python node to the HDL32 for time sync. GX5 is generating the clock for the system.