TixiaoShan/LIO-SAM #226: Ouster + xsens imu local trajectory problem

minwoo0611, 2021-04-05:
Hello. I'm using Ouster 128 and xsens mti 300.
When I acquire data with the above sensor and execute LIO-SAM using it, there seems to be a problem with the local trajectory as shown in the following figure.

When rotating, it does not seem to be smooth and it seems to vibrate, and even when it is simply going straight, the following left-right shaking occurs.

If anyone has ever had a similar experience, please advise.

Thank you.

[image]

[image]

horverno, 2021-04-11:
Check whether `imuGravity `differs from 9.8, i had similar issues, after averaging the IMU raw data it turned out it was significantly larger

closed 2021-06-03 by stale[bot]

learnhard22, 2022-08-21:
>  Also, I synced lidar time and imu time with ROS_TIME. 

Hello @minwoo0611,

I'm have problems to sync the Ouster OS0-128 with my external IMU (EllipseD - SBG Systems) and I want to ask you how you did time synchronization of the lidar with ROS time? The external IMU already uses ROS time, but not the Ouster.

Sorry in advantage, my issue is not directly connected with yours. Hopefully you can give me some advice to solve it.
I'm very new new into this Topic.

Best regards,
Lennard