TixiaoShan/LIO-SAM #227: Ouster + Brick v2 IMU extrinsics setup

Jaetriel, 2021-04-07:
Hello, I've got a setup consisting of the OS1-128 lidar and a Brick v2 IMU, a 9-axis external IMU. I have managed to get the configuration up and running, orientation seems mostly fine, but the problem is if I start moving the sensors a bit faster, or sometimes just on their own, I start getting jerking and even bigger jumps and leaps and the whole setup becomes pretty much unstable and unusable. 
[image]

[image]

[image]

I have managed to sync the timestamps correctly(as far as i have tested and know) using PTP and the lidar is running in TIME_FROM_PTP_1588 mode. Could this be also caused by the fact that the IMU's max frequency is only 100hz or is it just the case that the extrinsics are not setup correctly? My params.yaml file looks like this: 
`lio_sam:

  # Topics
  pointCloudTopic: "/os_cloud_node/points"               # Point cloud data
  imuTopic: "/tfsensors/imu1"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "os_sensor"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useImuHeadingInitialization: true           # if using GPS data, set to "true"
  useGpsElevation: false                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data
  
  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/Downloads/LOAM/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

  # Sensor Settings
  sensor: ouster                              # lidar sensor type, either 'velodyne' or 'ouster'
  N_SCAN: 128                                 # number of lidar channel (i.e., 16, 32, 64, 128)
  Horizon_SCAN: 2048                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048)
  # maybe check this
  downsampleRate: 1                           # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1
  lidarMinRange: 1.0                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 1000.0                       # default: 1000.0, maximum lidar range to be used

  # IMU Settings
  #imuAccNoise: 3.9939570888238808e-03
  #imuGyrNoise: 1.5636343949698187e-03
  #imuAccBiasN: 6.4356659353532566e-05
  #imuGyrBiasN: 3.5640318696367613e-05
  #imuGravity: 9.80511
  #imuRPYWeight: 0.01

  imuAccNoise: 3.9939570888238808e-03
  imuGyrNoise: 1.5636343949698187e-03
  imuAccBiasN: 6.4356659353532566e-05
  imuGyrBiasN: 3.5640318696367613e-05
  imuGravity: 9.80511
  imuRPYWeight: 0.01

  #result euler angle(RPY) : -0.0510692 -0.037892 0.179439
  #result extrinsic rotation matrix : 
  #0.983238  -0.176342 -0.0463368
  #0.178349   0.983006   0.043475
  #0.0378829 -0.0510103   0.997979

  # Extrinsics (lidar -> IMU)
  extrinsicTrans: [0.0, 0.0, 0.0]
  extrinsicRot: [1, 0, 0,
                 0, 1, 0,
                 0, 0, 1]
  extrinsicRPY: [1, 0, 0,
                 0, 1, 0,
                 0, 0, 1]
[... 90 lines cut ...]
                 false, false, false]
  odom0_differential: false
  odom0_queue_size: 10

  #                            x     y     z     r     p     y   x_dot  y_dot  z_dot  r_dot p_dot y_dot x_ddot y_ddot z_ddot
  process_noise_covariance: [  1.0,  0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    1.0,  0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    10.0, 0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0.03, 0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0.03, 0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0.1,  0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0.25,  0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0.25,  0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0.04,  0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0.01, 0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0.01, 0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0.5,  0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0.01, 0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0.01,   0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0.015]
`

I tried using the imu lidar calibration package in the project and the results I've put commented in the yaml file and I tried replacing the extrinsicRot and extrinsicRPY values with the ones I got from the package, but if anything, the situation got a little bit worse and I'm not exactly sure how to proceed to get the correct extrinsics for my setup?  

Any help or suggestions would be appreciated, thanks.

Jaetriel, 2021-04-12:
Update: So the biggest issue was a little oversight on my part, in my IMU node it was taking linear acceleration instead of acceleration with the effects of gravity accounted for, so now it's a lot more stable. One issue still remains, that if I rotate the sensor a bit more rapidly, or if it stays rotated for a longer period of time, it gets some weird acceleration values and starts roaming again, could this be some sort of wrong extrinsic or gravity values?

closed 2021-05-20 by Jaetriel

ataparlar, 2022-02-17:
Hi there,
Could you solve the roaming problem after configured your IMU? I am dealing with the same issue. Map occurs correct first but after moving about 10 meters, it starts to roaming. 
Have a good day.
ataparlar

18954534983, 2022-09-19:
> Hi there, Could you solve the roaming problem after configured your IMU? I am dealing with the same issue. Map occurs correct first but after moving about 10 meters, it starts to roaming. Have a good day. ataparlar

> Hello, I have encountered the same problem as you, but the method of calibration and time stamp synchronization has not been solved. May I ask how you solved this problem?

18954534983, 2022-09-19:
> Hi there, Could you solve the roaming problem after configured your IMU? I am dealing with the same issue. Map occurs correct first but after moving about 10 meters, it starts to roaming. Have a good day. ataparlar

Hello, I have encountered the same problem as you. Since my IMU position is above the ridar, I ruled out the reason that the external parameter calibration of IMU and ridar is not accurate. I synchronized the time stamps of IMU and ridar, but the problem of ground rise still exists. How did you solve it? Thank you.