TixiaoShan/LIO-SAM #229: wont compile

bobusfil, 2021-04-17:
Hello, I do not know what could be the problem I installed all the dependencies but when I try to compile the workspace with 
catkin build I get this:

Warnings   << lio_sam:cmake /home/admin-vras/lio_ws/logs/lio_sam/build.cmake.000.log                                                                                                                              
CMake Warning (dev) at CMakeLists.txt:101 (add_executable):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "lio_sam_imuPreintegration" links to target "Boost::timer" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:95 (add_executable):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "lio_sam_mapOptmization" links to target "Boost::timer" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/admin-vras/lio_ws/build/lio_sam; catkin build --get-env lio_sam | catkin env -si  /usr/bin/cmake /home/admin-vras/lio_ws/src/LIO-SAM --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/admin-vras/lio_ws/devel/.private/lio_sam -DCMAKE_INSTALL_PREFIX=/home/admin-vras/lio_ws/install; cd -
..................................................................................................................................................................................................................
__________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << lio_sam:make /home/admin-vras/lio_ws/logs/lio_sam/build.make.000.log                                                                                                                                
/home/admin-vras/lio_ws/src/LIO-SAM/src/imuPreintegration.cpp: In constructor ‘TransformFusion::TransformFusion()’:
/home/admin-vras/lio_ws/src/LIO-SAM/src/imuPreintegration.cpp:53:43: warning: catching polymorphic type ‘tf::TransformException’ {aka ‘class tf2::TransformException’} by value [-Wcatch-value=]
             catch (tf::TransformException ex)
                                           ^~
/home/admin-vras/lio_ws/src/LIO-SAM/src/mapOptmization.cpp: In member function ‘bool mapOptimization::saveMapService(lio_sam::save_mapRequest&, lio_sam::save_mapResponse&)’:
/home/admin-vras/lio_ws/src/LIO-SAM/src/mapOptmization.cpp:361:11: warning: variable ‘unused’ set but not used [-Wunused-but-set-variable]
       int unused = system((std::string("exec rm -r ") + saveMapDirectory).c_str());
           ^~~~~~
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/include/utility.h:26,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/src/featureExtraction.cpp:1:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/include/utility.h:26,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/include/utility.h:26,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/src/imuPreintegration.cpp:1:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/include/utility.h:26,
                 from /home/admin-vras/lio_ws/src/LIO-SAM/src/mapOptmization.cpp:1:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
/usr/bin/ld: cannot find -lBoost::timer
collect2: error: ld returned 1 exit status
make[2]: *** [/home/admin-vras/lio_ws/devel/.private/lio_sam/lib/lio_sam/lio_sam_imuPreintegration] Error 1
make[1]: *** [CMakeFiles/lio_sam_imuPreintegration.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: cannot find -lBoost::timer
collect2: error: ld returned 1 exit status
make[2]: *** [/home/admin-vras/lio_ws/devel/.private/lio_sam/lib/lio_sam/lio_sam_mapOptmization] Error 1
make[1]: *** [CMakeFiles/lio_sam_mapOptmization.dir/all] Error 2
make: *** [all] Error 2

yst1, 2021-04-25:
hello, i get the same problem, how do you solve this

bobusfil, 2021-04-25:
Hello, I added in to CMakeLists.txt:
cmake_policy(SET CMP0028 NEW)  - that takes care of the cmake warning about the policy.
Then

find_package(Boost REQUIRED COMPONENTS timer)

and after given add_executable i added
${Boost_LIBRARIES} to the list of libraries and 
target_link_libraries(${PROJECT_NAME}_mapOptmization Boost::timer)

Basically I followed examples in:
https://cmake.org/cmake/help/latest/module/FindBoost.html

yst1, 2021-04-26:
Sincerely for your kind help. Wish you a nice day!

Mikor-mkr, 2021-05-06:
@bobusfil  can you elaborate please, I am very new to this. Thanks in advance.

LukeAI, 2021-05-22:
any ideas on how to deal with:

```
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52,
from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
from /home/admin-vras/lio_ws/src/LIO-SAM/include/utility.h:26,
from /home/admin-vras/lio_ws/src/LIO-SAM/src/featureExtraction.cpp:1:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52,
from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
from /home/admin-vras/lio_ws/src/LIO-SAM/include/utility.h:26,
from /home/admin-vras/lio_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
^~~~~~~~~~~~~~~
```

closed 2021-06-09 by stale[bot]

KleinYuan, 2022-02-14:
For whoever met this issue, the following fixes on my end:

Add the following in [here](https://github.com/TixiaoShan/LIO-SAM/blob/master/CMakeLists.txt#L7)
```
find_package(Boost REQUIRED COMPONENTS timer)
```
Add the following at the end, after [here](https://github.com/TixiaoShan/LIO-SAM/blob/master/CMakeLists.txt#L102)
```
# Boost
target_link_libraries(${PROJECT_NAME}_mapOptmization Boost::timer)
```