TixiaoShan/LIO-SAM #230: double free or corruption (out)

gyjun0230, 2021-04-19:
Hi ,
Im sorry to trouble you, but I have followed the step for installation but I encountered the follow problem and Im not sure how to fix it. Can you give me some advice?

[1mROS_MASTER_URI=http://local*** Error in `/home/gaoyajun/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration': double free or corruption (out): 0x00000000015006a0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777f5)[0x7f1f31be17f5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8038a)[0x7f1f31bea38a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f1f31bee58c]
/usr/local/lib/libgtsam.so.4(_ZN5gtsam10noiseModel8Diagonal6SigmasERKN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEEb+0x173)[0x7f1f5523ef33]
/usr/local/lib/libgtsam.so.4(+0x109545)[0x7f1f550f2545]
/lib64/ld-linux-x86-64.so.2(+0x106fa)[0x7f1f6b4786fa]
/lib64/ld-linux-x86-64.so.2(+0x1080b)[0x7f1f6b47880b]
/lib64/ld-linux-x86-64.so.2(+0xc6a)[0x7f1f6b468c6a]
======= Memory map: ========
00400000-00473000 r-xp 00000000 103:02 45432305                          /home/gaoyajun/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration
00673000-00678000 rw-p 00073000 103:02 45432305                          /home/gaoyajun/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration
01455000-01514000 rw-p 00000000 00:00 0                                  [heap]
7f1f0c000000-7f1f0c021000 rw-p 00000000 00:00 0
7f1f0c021000-7f1f10000000 ---p 00000000 00:00 0
7f1f117cc000-7f1f11a37000 rw-p 00000000 00:00 0
7f1f11a37000-7f1f11a75000 r-xp 00000000 103:02 41812106                  /usr/lib/x86_64-linux-gnu/libquadmath.so.0.0.0
7f1f11a75000-7f1f11c74000 ---p 0003e000 103:02 41812106                  /usr/lib/x86_64-linux-gnu/libquadmath.so.0.0.0

closed 2021-06-03 by stale[bot]

jhch1995, 2021-09-02:
@gyjun0230 I have the same problem, can you tell me the way to solve this problem?

HaisenbergPeng, 2021-09-28:
try pcl1.7 instead of pcl1.8

MaybeShewill-CV, 2021-11-03:
@gyjun0230 @jhch1995 @HaisenbergPeng Have you guys solve the problem? May i ask you for your help? Thanks:)

zhang275, 2021-12-02:
I think it conflicts with the PCL

chxyfishge, 2024-07-29:
Follow @zhang275 
I have the same question，and I soved it. Let me explain my situation, maybe it would be helpful.
Ubuntu 20.04 provided PCL 1.10.0 in /usr/add and/usr/lib, but I compiled another PCL 1.10.0 and installed it in/usr/local/. The conflict between these two PCLs resulted in
double free or corruption (out),
All we need to do is uninstall the compiled PCL.