TixiaoShan/LIO-SAM #231: Failed to save map

ghz009, 2021-04-20:
Thank you for your code, which helped me a lot.
And when I play back my recorded bag and want to save the map, I get the following error:

terminate called after throwing an instance of 'pcl::IOException'
  what():  : [pcl::PCDWriter::writeBinary] Input point cloud has no data!

ibrahimhroob, 2021-04-27:
Hi @ghz009

Have you figured out how to save the map/path? 

Thanks

ghz009, 2021-04-28:
> Hi @ghz009
> 
> Have you figured out how to save the map/path?
> 
> Thanks

Thank you for your reply.
I know how to save the map and can successfully save the official map.
But when I save my own map, I get errors.

christophTUM, 2021-05-27:
> > Hi @ghz009
> > Have you figured out how to save the map/path?
> > Thanks
> 
> Thank you for your reply.
> I know how to save the map and can successfully save the official map.
> But when I save my own map, I get errors.

How did you save the official map?

ibrahimhroob, 2021-05-27:
In the params.yaml file, you need to set "savePCD" to true

closed 2021-06-16 by stale[bot]