TixiaoShan/LIO-SAM #232: Image projection keeps restarting!

ibrahimhroob, 2021-04-20:
Hi @TixiaoShan 

I have an issue when running the algorithm, not sure why but the image projection keeps restarting and I wan not able to debug the source if the issue. I have attached the terminal output messages. Any help would be much appreciated, many thanks. 

[image]

ibrahimhroob, 2021-04-23:
The issue was with the data from velodyne, for some reason the point cloud width and height were the other way around, swapping them back seems to fix the issue!

closed 2021-04-23 by ibrahimhroob

lyhbit, 2021-04-23:
very good

jyothishkiran, 2021-05-19:
hi @ibrahimhroob i am new to autonomous projects. I have the same issue while running the algorithm. Could you tell me how you swapped height and width of point cloud?

ibrahimhroob, 2021-05-19:
Hi @jyothishkiran 
In the image projection node modified the first bit of `cachePointCloud` function to the following:
`

    bool cachePointCloud(const sensor_msgs::PointCloud2ConstPtr& laserCloudMsg)
    {
        // cache point cloud
        cloudQueue.push_back(*laserCloudMsg);
        if (cloudQueue.size() <= 2)
            return false;

        // convert cloud
        currentCloudMsg = std::move(cloudQueue.front());
        cloudQueue.pop_front();

        //For some reason the point cloud data from the velodyne plugin does not follow stander point cloud data structure!!   
        if(currentCloudMsg.height > 1){
            currentCloudMsg.width = currentCloudMsg.height;
            currentCloudMsg.height = 1;
        }
        currentCloudMsg.header.frame_id = "base_link";

.
.
.

` 

Hope it helps

jyothishkiran, 2021-05-19:
thank you so much @ibrahimhroob  it worked

cs2514, 2021-05-20:
hi @ibrahimhroob i have an issue in generated map. could you help me solve this https://github.com/TixiaoShan/LIO-SAM/issues/244#issue-895805726