TixiaoShan/LIO-SAM #233: Export the generated path

ibrahimhroob, 2021-04-27:
Hi @TixiaoShan 

How I can export the generated trajectory?

ibrahimhroob, 2021-05-03:
@TixiaoShan , is there a method to save the trajectory in tum format!?

closed 2021-06-02 by ibrahimhroob

LongruiDong, 2022-03-24:
Hi

> @TixiaoShan , is there a method to save the trajectory in tum format!?

Hi~

have you figured out how to save final optimized pose?

ibrahimhroob, 2022-03-24:
yes, I was saving lio-sam output pose in rosbag, then I used evo library to plot it and save it into tum format (https://github.com/MichaelGrupp/evo)

LongruiDong, 2022-03-26:
> yes, I was saving lio-sam output pose in rosbag, then I used evo library to plot it and save it into tum format (https://github.com/MichaelGrupp/evo)

Thans for your reply, could you explain more specifically that how to save lio-sam output pose in rosbag?

ibrahimhroob, 2022-03-26:
Hi, yeah no problem we have all been there! so after launching lio-sam you need to check to published topics, I think lio-sam publishes two odometry topics that are:
* Global odometry: "lio_sam/mapping/odometry"
* Incremental odometry: "lio_sam/mapping/odometry_incremental"

To record those topics into rosbag you simply type: 
`rosbag record -O liosam_pose.bag lio_sam/mapping/odometry lio_sam/mapping/odometry_incremental`

For plotting the trajectory using evo, you can use the following line: 
`evo_traj bag liosam_pose.bag lio_sam/mapping/odometry_incremental lio_sam/mapping/odometry -p --plot_mode xy`

hope this helps!