TixiaoShan/LIO-SAM #238: Point clouds are rotating and jerking when sensors rotating

fergcd, 2021-05-07:
Point clouds generated by LOAM (IMU not used)  :
[image]
[image]
Looks fine in a top view, but roads are distorted in a side side view.

So I decided to use LIO-SAM to minimize the distortion.
Before testing outside, I ran LIO-SAM with VLP-32C(32ch LiDAR) and 9-axis IMU in my office, and it looks like it's working. (See #234)
(IMU's link : https://www.devicemart.co.kr/goods/view?no=1310790)
[image]

Point clouds generated by LIO-SAM (9-axis IMU and VLP-32C mounted on the car)
[image]
But when tested outside, point clouds are rotating and jerking when sensors rotating!
In fact, if I don't set IMU's linear acceleration and angular velocity to fixed values like below, it gets worse. (Terrible zigzag and jerking)

`sensor_msgs::Imu imuConverter(const sensor_msgs::Imu& imu_in)
    {
        sensor_msgs::Imu imu_out = imu_in;
        // rotate acceleration
        //Eigen::Vector3d acc(imu_in.linear_acceleration.x, imu_in.linear_acceleration.y, imu_in.linear_acceleration.z);
        Eigen::Vector3d acc(0.0, 0.0, -9.80665);  // use fixed linear acceleration
        acc = extRot * acc;
        imu_out.linear_acceleration.x = acc.x();
        imu_out.linear_acceleration.y = acc.y();
        imu_out.linear_acceleration.z = acc.z();
        // rotate gyroscope
        //Eigen::Vector3d gyr(imu_in.angular_velocity.x, imu_in.angular_velocity.y, imu_in.angular_velocity.z);
        Eigen::Vector3d gyr(0.0, 0.0, 0.0); // use fixed angular velocity
        gyr = extRot * gyr;
        imu_out.angular_velocity.x = gyr.x();
        imu_out.angular_velocity.y = gyr.y();
        imu_out.angular_velocity.z = gyr.z();
        // rotate roll pitch yaw
        Eigen::Quaterniond q_from(imu_in.orientation.w, imu_in.orientation.x, imu_in.orientation.y, imu_in.orientation.z);
        Eigen::Quaterniond q_final = q_from * extQRPY;
        imu_out.orientation.x = q_final.x();
        imu_out.orientation.y = q_final.y();
        imu_out.orientation.z = q_final.z();
        imu_out.orientation.w = q_final.w();

        if (sqrt(q_final.x()*q_final.x() + q_final.y()*q_final.y() + q_final.z()*q_final.z() + q_final.w()*q_final.w()) < 0.1)
        {
            ROS_ERROR("Invalid quaternion, please use a 9-axis IMU!");
            ros::shutdown();
        }

        return imu_out;
    }`

My ROS bag file (11.8GB) :
https://drive.google.com/file/d/1tL_1Txjh8emt5DyuXcmK1TBd2JKCQTNA/view?usp=sharing

Even if I tried commented extrinsicRot and extrinsicRPY in params.yaml, it didn't work.
My params.yaml :

`lio_sam:

  # Topics
  pointCloudTopic: "velodyne_points"               # Point cloud data
  imuTopic: "imu_raw"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
[... 123 lines cut ...]
                 false, false, false]
  odom0_differential: false
  odom0_queue_size: 10

  #                            x     y     z     r     p     y   x_dot  y_dot  z_dot  r_dot p_dot y_dot x_ddot y_ddot z_ddot
  process_noise_covariance: [  1.0,  0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    1.0,  0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    10.0, 0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0.03, 0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0.03, 0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0.1,  0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0.25,  0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0.25,  0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0.04,  0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0.01, 0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0.01, 0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0.5,  0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0.01, 0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0.01,   0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0.015]
`

I think my IMU's linear acceleration and angular velocity have a big noise, so I'm trying to apply Low Pass Filter or Kalman Filter to them to minimize the noise.

Are there other things to try to solve this issue?

lincob, 2021-05-07:
1. Calibrate the IMU using the procedure described somewhere.
2. Buy a higher grade IMU
3. Try with Lego loam

fergcd, 2021-05-07:
@lincob Thank you for fast answer!  I'll try.

lincob, 2021-05-19:
@fergcd I've just redone the calibration again in better conditions and the improvement in terms of jerking is quite noticeable. It's worth the try.
My problem was that the HDL32e I'm using is quite a beefy rotating piece and creates a lot of vibrations. I realized that even mounted on a padded platform, my imu was probably too close to the Lidar unit.
Now I'm still running into timing problems but that's another story.
Anyway, the package in question is :
https://github.com/gaowenliang/imu_utils
I've ran a 130min static session with lidar spinning. (Don't be like me, just record IMU data and not ALL topics 😉 otherwise you may end up with a VERY large file 🤣)

fergcd, 2021-05-20:
@lincob Thank you, I'm touched by your kind comment!!  I'll also try that imu utils you shared!

closed 2021-06-06 by stale[bot]