TixiaoShan/LIO-SAM #244: Map appearing in red color only

jyothishkiran, 2021-05-19:
Hi @TixiaoShan i am a newbie and i was doing mapping using LIO SAM. Everything worked perfectly without any errors when i used my bag but I am getting map only in red color(rviz dispaly). I dont know whether this is an issue since all the other bags(eg: campus dataset) produced colourful maps. 
[image]
[image]
[image]
[image]

jyothishkiran, 2021-05-19:
> Hi @TixiaoShan i am a newbie and i was doing mapping using LIO SAM. Everything worked perfectly without any errors when i used my bag but I am getting map only in red color(rviz dispaly). I dont know whether this is an issue since all the other bags(eg: campus dataset) produced colourful maps.
> [image]
> [image]
> [image]
> ![7EA2530B-0A4A-498B-8BE4-EC46058AE9B8](https://user-images.githubusercontent.com/83896005/118872694-e697af00-b8f9-11eb-8186-d69c1c9d39a5

cbaus, 2021-05-20:
Do you see a coloured map when you switch on `Map (Global)` in rviz?

cs2514, 2021-05-20:
hi @cbaus i tried switching on Map (Global) in rviz but still it is in red color. i will attach the screen shots of my configurations.
/home/jyothish/Pictures/Screenshot from 2021-05-20 15-27-36.png
/home/jyothish/Pictures/Screenshot from 2021-05-20 15-27-44.png

i launched campus dataset bag using LIO SAM but i got colored map with same rviz configuration
/home/jyothish/Pictures/Screenshot from 2021-05-20 15-33-03.png

cs2514, 2021-05-20:
hi @cbaus i tried switching on Map (Global) in rviz but still it is in red color. i will attach the screen shots of my configurations.
[image]
[image]

i launched campus dataset bag using LIO SAM but i got colored map with same rviz configuration
[image]

cbaus, 2021-05-20:
I see. I am really new at using lio-sam but maybe check if your point clouds have intensity value (XYZ**I**)

lucksonzhen, 2021-05-21:
@cs2514 Hi, could you please show me your config yaml for campus dataset? Thanks a lot!!! I think I have followed the guide from the readme.md , and tried a lot of change to the parameters, but the mapping process for 3 datasets I've tried seems quite messy, with the warning of "Large velocity, reset imu preintegration"&"not enough features" a lot.

cs2514, 2021-05-21:
@lucksonzhen sure 

lio_sam:

  # Topics
  pointCloudTopic: "points_raw"               # Point cloud data
  imuTopic: "imu_correct"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useImuHeadingInitialization: false          # if using GPS data, set to "true"
  useGpsElevation: false                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data
  
  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/Downloads/LOAM/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

  # Sensor Settings
  sensor: velodyne                            # lidar sensor type, either 'velodyne' or 'ouster'
  N_SCAN:   16                              # number of lidar channel (i.e., 16, 32, 64, 128)
  Horizon_SCAN: 1800                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048)
  timeField: "time" 
  downsampleRate: 1                           # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1
  lidarMinRange: 1.0                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 1000.0                       # default: 1000.0, maximum lidar range to be used

  # IMU Settings
  imuAccNoise: 3.9939570888238808e-03
  imuGyrNoise: 1.5636343949698187e-03
  imuAccBiasN: 6.4356659353532566e-05
  imuGyrBiasN: 3.5640318696367613e-05
  imuGravity: 9.80511
  imuRPYWeight: 0.01

  # Extrinsics (lidar -> IMU)
  extrinsicTrans: [0, 0, 0]
  extrinsicRot: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]
  extrinsicRPY: [1,  0, 0,
                 0, 1, 0,
                  0, 0, 1]
  # extrinsicRot: [1, 0, 0,
  #                 0, 1, 0,
  #                 0, 0, 1]
  # extrinsicRPY: [1, 0, 0,
  #                 0, 1, 0,
  #                 0, 0, 1]

  # LOAM feature threshold
  edgeThreshold: 1.0
  surfThreshold: 0.1
  edgeFeatureMinValidNum: 10
  surfFeatureMinValidNum: 100

  # voxel filter paprams
  odometrySurfLeafSize: 0.4                     # default: 0.4 - outdoor, 0.2 - indoor
  mappingCornerLeafSize: 0.2                    # default: 0.2 - outdoor, 0.1 - indoor
  mappingSurfLeafSize: 0.4                      # default: 0.4 - outdoor, 0.2 - indoor

  # robot motion constraint (in case you are using a 2D robot)
[... 66 lines cut ...]
  odom0: odometry/gps
  odom0_config: [true,  true,  true,
                 false, false, false,
                 false, false, false,
                 false, false, false,
                 false, false, false]
  odom0_differential: false
  odom0_queue_size: 10

  #                            x     y     z     r     p     y   x_dot  y_dot  z_dot  r_dot p_dot y_dot x_ddot y_ddot z_ddot
  process_noise_covariance: [  1.0,  0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    1.0,  0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    10.0, 0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0.03, 0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0.03, 0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0.1,  0,     0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0.25,  0,     0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0.25,  0,     0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0.04,  0,    0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0.01, 0,    0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0.01, 0,    0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0.5,  0,    0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0.01, 0,      0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0.01,   0,
                               0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0.015]

cs2514, 2021-05-21:
@lucksonzhen sure
[params.pdf](https://github.com/TixiaoShan/LIO-SAM/files/6520727/params.pdf)

check lidar -> IMU transform part in the above file.

cs2514, 2021-05-23:
hi @TixiaoShan  @cbaus i am using HDL-32e lidar. Tried many times changing lidar and the world but still the mapping process is coming in red color.

lucksonzhen, 2021-05-28:
> @lucksonzhen sure
> [params.pdf](https://github.com/TixiaoShan/LIO-SAM/files/6520727/params.pdf)
> 
> check lidar -> IMU transform part in the above file.

Thanks !

lincob, 2021-05-30:
> hi @TixiaoShan @cbaus i am using HDL-32e lidar. Tried many times changing lidar and the world but still the mapping process is coming in red color. it would be great if someone could help me on this.

What velodyne drivers are you using ? When you launch a rviz to visualize only the point cloud does it shows intensity ?

yxallnyy, 2021-06-10:
> @lucksonzhen sure
> [params.pdf](https://github.com/TixiaoShan/LIO-SAM/files/6520727/params.pdf)
> 
> check lidar -> IMU transform part in the above file.

hi,could you share your dataset & yaml, i get the same problem. thx

yxallnyy, 2021-06-10:
> Hi @TixiaoShan i am a newbie and i was doing mapping using LIO SAM. Everything worked perfectly without any errors when i used my bag but I am getting map only in red color(rviz dispaly). I dont [...]

> Hi @TixiaoShan i am a newbie and i was doing mapping using LIO SAM. Everything worked perfectly without any errors when i used my bag but I am getting map only in red color(rviz dispaly). I dont [...]

hi, could you share your data & yaml ? i got the same problem. thx

closed 2021-06-28 by stale[bot]

viettiepx2, 2025-02-10:
Hi @cs2514 !
I got same issue, did you fix it?