TixiaoShan/LIO-SAM #245: What is the correct setting for odomTopic for KITTI? (reset IMU-preintegration)

cbaus, 2021-05-20:
A couple of times in the issues i have seen the following parmaters for KITTI example:
```yaml
  # Topics
  pointCloudTopic: "/points_raw"               # Point cloud data
  imuTopic: "/imu_raw"                         # IMU data
  odomTopic: "/odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "/gps/fix"                   # GPS odometry topic from navsat, see module_navsat.launch file
```

However, the example KITTI bag and also custom produced do not have `/odometry/imu`. Available topics are:
```
topics:      /gps/fix        5177 msgs    : sensor_msgs/NavSatFix     
             /gps/vel        5177 msgs    : geometry_msgs/TwistStamped
             /imu_correct   53663 msgs    : sensor_msgs/Imu           
             /imu_raw       53663 msgs    : sensor_msgs/Imu           
             /points_raw     5177 msgs    : sensor_msgs/PointCloud2
```
The example bag runs though. The edges of the pointcloud are not chrystal clear though. With other bags (e.g. `2011-10-03 drive 0027`) I see jumps and error messages like:
`[ WARN] [1621520482.627370204]: Large velocity, reset IMU-preintegration!`
even when running at rate `0.5`.

Is the `odomTopic` setting correct and are there other suggestions to fix the `Large velocity` warning?

closed 2021-06-06 by stale[bot]

tejalbarnwal, 2021-12-02:
I am facing the same issue. Could someone please help?

LongruiDong, 2022-03-27:
> A couple of times in the issues i have seen the following parmaters for KITTI example:
> 
> ```yaml
>   # Topics
>   pointCloudTopic: "/points_raw"               # Point cloud data
>   imuTopic: "/imu_raw"                         # IMU data
>   odomTopic: "/odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
>   gpsTopic: "/gps/fix"                   # GPS odometry topic from navsat, see module_navsat.launch file
> ```
> 
> The example KITTI bag and also custom produced have the following topics:
> 
> ```
> topics:      /gps/fix        5177 msgs    : sensor_msgs/NavSatFix     
>              /gps/vel        5177 msgs    : geometry_msgs/TwistStamped
>              /imu_correct   53663 msgs    : sensor_msgs/Imu           
>              /imu_raw       53663 msgs    : sensor_msgs/Imu           
>              /points_raw     5177 msgs    : sensor_msgs/PointCloud2
> ```
> 
> `odomTopic` is only for publishing right? Should I use `imu_raw` or `imu_correct`? It seems the values are identical in the two topics.
> 
> No matter which I choose, the example bag runs. The edges of the pointcloud are not chrystal clear though. With other bags (e.g. `2011-10-03 drive 0027`) I see jumps and error messages like: `[ WARN] [1621520482.627370204]: Large velocity, reset IMU-preintegration!` even when playing the rosbag at rate `0.5` and no downsampling. The speed would continually increase:
> 
> ```
> Current speed=91.2445km/h  (v_z=-10.6621km/h)
> 
> Current speed=99.3167km/h  (v_z=-11.947km/h)
> 
> Current speed=102.449km/h  (v_z=-11.5976km/h)
> 
> Current speed=104.578km/h  (v_z=-11.4897km/h)
> 
> Current speed=105.283km/h  (v_z=-12.11km/h)
> 
> Current speed=106.828km/h  (v_z=-12.6063km/h)
> 
> Current speed=113.21km/h  (v_z=-15.7767km/h)
> ```
> 
> Are there other suggestions to fix the `Large velocity` warning? Could it be wrong IMU bias/noise values?

I am also using LIO-SAM to test this kitti sequences, showing this warning and the pose becomes wrong crazily, have you solved this problem?

LongruiDong, 2022-03-27:
by the way, do you add timestamp for each point in a cloud frame in kitti rosbag?

Sunshinehaozi, 2022-04-17:
Hi,I have meet the same problem. Does someone solve it?