TixiaoShan/LIO-SAM #247: ImuPreintegration and ImageProjection crashing and restarting

Jaetriel, 2021-05-25:
Hi, I've been trying to make this work with a Bosch BNO055 chip until now, but recently I managed to get a Microstrain 3DM-GX5-25, like @TixiaoShan is using, but after switching to it and changing the topics around, I've noticed I started getting a lot more errors and crashes of my system with output similar to this:

> terminate called after throwing an instance of 'std::runtime_error'
 what():  PreintegratedImuMeasurements::integrateMeasurement: dt <=0
[lio_sam_imuPreintegration-1] process has died [pid 574, exit code -6, cmd /home/gl/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/home/gl/.ros/log/1415f1e0-bd6d-11eb-b877-64bc5855aa9b/lio_sam_imuPreintegration-1.log].
log file: /home/gl/.ros/log/1415f1e0-bd6d-11eb-b877-64bc5855aa9b/lio_sam_imuPreintegration-1*.log
[lio_sam_imuPreintegration-1] restarting process
process[lio_sam_imuPreintegration-1]: started with pid [691]
[ INFO] [1621956396.629035437]: ----> IMU Preintegration Started.
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[lio_sam_imageProjection-2] process has died [pid 575, exit code -6, cmd /home/gl/catkin_ws/devel/lib/lio_sam/lio_sam_imageProjection __name:=lio_sam_imageProjection __log:=/home/gl/.ros/log/1415f1e0-bd6d-11eb-b877-64bc5855aa9b/lio_sam_imageProjection-2.log].
log file: /home/gl/.ros/log/1415f1e0-bd6d-11eb-b877-64bc5855aa9b/lio_sam_imageProjection-2*.log
[lio_sam_imageProjection-2] restarting process
process[lio_sam_imageProjection-2]: started with pid [717]
[ INFO] [1621956403.494835027]: ----> Image Projection Started.

I am using an Ouster OS1-128 lidar, does anyone have any suggestions how to fix these errors? thanks

JasperTan97, 2021-05-27:
I get this randomly too. I  fixed it yesterday but today the issue is back. Something is unstable with the code.

JasperTan97, 2021-05-27:
Hi, I think you may have different versions of GTSAM? Or something with gtsam is messing up. 
Repeat these steps

```
cd ~/Downloads/gtsam-4.0.2/
mkdir build && cd build
cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
sudo make install -j8
```

Then roslaunch the run file. It worked for me.

Jaetriel, 2021-05-27:
Hi @JasperTan97 , I think most of those issues I managed to resolve, but when I start rotating the sensor without even actually moving it just rotating in place, it starts drifting significantly, please see the attached video. Could this be cause by unsynced timestamps or possibly something else?

https://user-images.githubusercontent.com/43788971/119853691-b612cf00-bf10-11eb-9709-29edcd96d257.mp4

ajay1606, 2021-06-06:
@JasperTan97 Thanks for highlighting this issue, would be great if could share how you fixed this issue. Currently, am also stuck with the same problem.

**run. launch** resulting in the following errors:
```
[lio_sam_imuPreintegration-1] process has died [pid 6921, exit code 127, cmd /home/ajay/lio_sam_ws/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/home/ajay/.ros/log/3896f47a-c695-11eb-a00d-f832e4708d26/lio_sam_imuPreintegration-1.log].
log file: /home/ajay/.ros/log/3896f47a-c695-11eb-a00d-f832e4708d26/lio_sam_imuPreintegration-1*.log
[lio_sam_mapOptmization-4] process has died [pid 6922, exit code 127, cmd /home/ajay/lio_sam_ws/devel/lib/lio_sam/lio_sam_mapOptmization __name:=lio_sam_mapOptmization __log:=/home/ajay/.ros/log/3896f47a-c695-11eb-a00d-f832e4708d26/lio_sam_mapOptmization-4.log].
log file: /home/ajay/.ros/log/3896f47a-c695-11eb-a00d-f832e4708d26/lio_sam_mapOptmization-4*.log
```

Working Platform:
Ubuntu 18.04

Greatly appreciate your response,

 Kind regards,
Ajay

JasperTan97, 2021-06-06:
Hi @ajay1606 , what version of GTSAM are you using. For me, I used a different version of gtsam because legoloam required a different version. 
Just make sure you're using gtsam 4.0.2

ajay1606, 2021-06-06:
@JasperTan97 Thanks much for your quick response, I am also using gtsam 4.0.2 version and have followed the instruction as it mentioned.

Also, have repeated these steps again, but couldn't get rid of the error message. Is there any other fix?

Thanks

ajay1606, 2021-06-07:
@JasperTan97 Thanks, Issue has been fixed. The actual problem was location of  **ibmetis.so**.
It was a run time library, but its location was not in the expected directory when the application looking for it. 
While installing library **gtsam** by running command **sudo make install -j8**  the file **ibmetis.so** installed in the default location at **/usr/local/lib/**, but when we launch ros workspace, runtime library looking at location **/opt/ros/melodic/lib/**

By moving file **ibmetis.so** from **/usr/local/lib/** to **/opt/ros/melodic/lib/** fixed the issue.

Thanks

closed 2021-06-26 by stale[bot]

cuge1995, 2021-07-16:
> @JasperTan97 Thanks, Issue has been fixed. The actual problem was location of library file **libmetis.so**.
> It was a run time library, but its location was not in the expected directory when the application looking for it.
> While installing library **gtsam** by running command **sudo make install -j8** the file **libmetis.so** installed in the default location at **/usr/local/lib/**, but when we launch ros workspace, runtime library looking at location **/opt/ros/melodic/lib/**
> 
> By moving file **libmetis.so** from **/usr/local/lib/** to **/opt/ros/melodic/lib/** fixed the issue.
> 
> Thanks

It works for me

Amit10311, 2021-07-16:
Hello @Jaetriel @ajay1606  @TixiaoShan,

Thank you for this great package and issue support, I saw you're using ouster lidar, I would like to know which ouster pkg you used, and could you please share the changes you made in params.yaml.  

1. I would like to mention that I am using  Os1-32 with `https://github.com/ouster-lidar/ouster_example` it's working with lidar but not working with the` LIO-SAM`. 
2. Imu is debugging but no data processing of Lidar data. 

Can you please guide me on which ouster pkg is working in LIO_SAM pkg? or is thr other prblm. 

I look forward to hearing from you. Thank you for your valuable time.

Regards, 
Amit

ajay1606, 2021-07-16:
@Amit10311 WOuld be very helpful to understand the issue if you could post an actual error message.

Amit10311, 2021-07-16:
@ajay1606 It's not showing any error, In `rqt-grap` it's sowing everything connected 
* [rqt_graph image ](https://drive.google.com/file/d/146tsfV4uKm9he4X-ttYqy1nQ1Ak2muNa/view?usp=sharing)
 but it's now processing poincloud data. 

*  In rviz no `map` frame, i tried to echo `deskew/cloud_deskewed` ,` lio_sam/deskew/cloud_info` no data publishing.

Jaetriel, 2021-07-16:
@Amit10311 Have you maybe checked the point cloud topics and made sure they are correct? The Ouster default example package published a topic called `/os_cloud_node/points` and LIO-SAM by default is looking for a topic called `points_raw` I believe. If you use the command `rostopic list` do you see the points topic by Ouster being published maybe you can echo that one and see if it's correct?

Amit10311, 2021-07-16:
@Jaetriel thanks for sugguestion, 

**Rosbag or velodyne** 
```
rostopic info /points_raw 

Type: sensor_msgs/PointCloud2

Publishers: None

Subscribers: 
 * /lio_sam_imageProjection
```
**Ouster** 
```
rostopic info  /os_cloud_node/points 

Type: sensor_msgs/PointCloud2

Publishers: 
 * /os_cloud_node 

Subscribers: 
 * /lio_sam_imageProjection
```
I also did `timestamp_mode` in your Ouster launch file to `TIME_FROM_PTP_1588`

No change, Are you using ouster1 with `LIO_SAM` pkg? 
Which ouster pkg did you use to run Ouster Lidar?

Amit10311, 2021-07-16:
@Jaetriel Thanks, It's working now. I just had to echo that one and see it's correct. Now I have to work on imu debugging. It's amazing.

Jaetriel, 2021-07-16:
@Amit10311 I use an OS1-128 sensor with the package that they have listed on their website https://github.com/ouster-lidar/ouster_example . It's a bit weird to me that both `points_raw` topic and `os_cloud_node/points` are subscribed by imageProjection. In LIO-SAM's `params.yaml` file you have switched the `pointCloudTopic` from `/points_raw` to `/os_cloud_node/points` right?

Jaetriel, 2021-07-16:
@Amit10311 Ah, okay, good to hear that you have managed to resolve it! Let me know if you still have some issues I will do my best to help, with my limited knowledge of the codebase. Cheers!

kartiksoni01, 2024-03-20:
Hey @Jaetriel great work!
I am trying to perform the similar configuration on my side by using a velodyne VLP16 along with Bosch BNO055 IMU. I wanted to know what imu package are you using to get the imu data. I am uing the ros_imu_bno055 package to take the data via USB, i feel it has a lot of noise, resulting into this
[image]
Can you guide me as to what direction i should proceed in to solve this issue, Thanks

Amit10311, 2024-05-28:
Use  PTP for  LiDAR, lidar and imu msg are not sysnc together. 
`sudo ptpd -i interface_name -M -V
`

sagar0609nagdev, 2024-05-29:
> @sagar0609nagdev Use PTP for LiDAR, lidar and imu msg are not sysnc together. `sudo ptpd -i interface_name -M -V `

I tried PTP but error is same. I mean bno055 is using serial interface so only ouster is started using PTP. 
Is there any other thing I am missing? or is this happening because my imu is not fully calibrated.

Amit10311, 2024-05-29:
@sagar0609nagdev I saw in the terminal your IMU is not calibrated. It's showing a warning check it.