TixiaoShan/LIO-SAM #253: Relocalization

adamanov, 2021-06-09:
Hi, 

First of all, thank you for your great work and contribution. 
After generating a successfully 3D scan map, can we re-use it for the relocalization of the robot in that environment? 
If yes, what kind of change do I need to apply?

All Best

closed 2021-06-28 by stale[bot]

Sollimann, 2022-02-08:
@adamanov I curious of the same thing. Did you figure out whether or not it was possible to do re-localization in existing map?

adamanov, 2022-02-09:
> @adamanov I curious of the same thing. Did you figure out whether or not it was possible to do re-localization in existing map?

unfortunately, I could not figure it out. But according to the paper, it was possible

Sollimann, 2022-02-09:
Okey, thank you. Which onecdid you go with? Mabye I could consider changing our SLAM solution as well

Sollimann, 2022-02-09:
> > @adamanov I curious of the same thing. Did you figure out whether or not it was possible to do re-localization in existing map?
> 
> 
> 
> unfortunately, I could not figure it out. I went with another slam approach 

Okey, thank you. Which onecdid you go with? Mabye I could consider changing our SLAM solution as well

adamanov, 2022-02-09:
We went with Cartographer, since it is well documented and has integration with ROS

adamanov, 2022-02-09:
> > > @adamanov I curious of the same thing. Did you figure out whether or not it was possible to do re-localization in existing map?
> > 
> > 
> > 
> > unfortunately, I could not figure it out. I went with another slam approach 
> 
> Okey, thank you. Which one did you go with? Mabye I could consider changing our SLAM solution as well

We went with Cartographer, since it is well documented and has integration with ROS