TixiaoShan/LIO-SAM #263: undefined reference to `void pcl::removeNaNFromPointCloud<VelodynePointXYZIRT>(pcl::PointCloud<VelodynePointXYZIRT> const&, pcl::PointCloud<VelodynePointXYZIRT>&, std::vector<int, std::allocator<int> >&)'

qserdel, 2021-06-28:
Hi,
I'm trying to use LIO-SAM with a Velodyne VLP-32 Lidar which seems to have NaN values.
I tried to remove them using :

`        vector<int> indices;`
`        pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);`

but get the following error :

``undefined reference to `void pcl::removeNaNFromPointCloud<VelodynePointXYZIRT>(pcl::PointCloud<VelodynePointXYZIRT> const&, pcl::PointCloud<VelodynePointXYZIRT>&, std::vector<int, std::allocator<int> >&)     ``

Please help me I don't know what to do

closed 2021-07-14 by stale[bot]

beyondpzk, 2021-09-13:
> Hi,
> I'm trying to use LIO-SAM with a Velodyne VLP-32 Lidar which seems to have NaN values.
> I tried to remove them using :
>
> ` vector<int> indices;`
> ` pcl::removeNaNFromPointCloud(*laserCloudIn, [...]

Have you solved this problem? I also have met this problem.

z1047941150, 2021-12-27:
Have you solved this problem? I have met this problem too.

flashlau, 2022-06-30:
`
undefined reference to `void pcl::removeNaNFromPointCloud<RsPointXYZIRT>(pcl::PointCloud<RsPointXYZIRT> const&, pcl::PointCloud<RsPointXYZIRT>&, std::vector<int, std::allocator<int> >&)'
`
It was solved  by adding #include "pcl-1.8/pcl/filters/impl/filter.hpp"

FALCONYU, 2023-04-18:
It was solved by adding #include "pcl-1.13/pcl/filters/impl/filter.hpp"