TixiaoShan/LIO-SAM #264: A question about function imuConverter()

Psyclonus2887, 2021-06-30:
First thank you for your excellent work! I apply LIO_SAM in our data set constructed by Hesai lidar and OxT GNSS system, and it performs well. But I have a question when reading the code.
In utility.h, the imuConverter() function is aimed to convert acc, gyr and RPY gained in IMU frame into lidar frame, in my opinion. But if so, RPY in lidar frame should be calculated by:
q_final = extQRPY * q_from * extQRPY_inv
Is there something wrong in this calculiation?

Psyclonus2887, 2021-06-30:
I refer to that because in one of my cars, the rotation matrix from lidar frame to imu frame is [0, -1, 0, -1, 0, 0, 0, 0, -1], but the transformed RPY is really strange.

closed 2021-07-20 by stale[bot]

narutojxl, 2021-11-16:
I agree with you @Psyclonus2887,  `q_final = extQRPY * q_from * extQRPY.inv()`.

hanhaoguang, 2022-02-09:
> * ptmization.cpp when

I agree with you. I think it should be q_final = q_from * extQRPY_inv