TixiaoShan/LIO-SAM #266: Guidance for using LIO-SAM for pure localization purpose

HappySamuel, 2021-07-04:
Hi everyone,

LIO-SAM is capable of creating 3D map with only 2 sensors (3D LiDAR and IMU). @TixiaoShan , your work is amazing.

This amazing work lead me to think if there's possibility to use LIO-SAM purely for localization purpose? And what kind of changes do i need to make? 

Concept:
(1) Use LIO-SAM to create 3D map, which also providing tf (map->odom & odom->base_link) and together with hector_mapping that listen to the tf and generating 2D map..
(2) Save the created 3D and 2D maps.
(3) Use localization edition of LIO-SAM for localizing in the 3D map just created, to navigate robot.

So the key things is which part of LIO-SAM can be modified to achieve (3). Need some guidance from author @TixiaoShan  and everyone.

Best,
Samuel

Huguet57, 2021-07-04:
Hi Samuel,
Not LIO-SAM but I managed to do that with A-LOAM. Basically what I did is save the 3D map in a .txt so then I could load it. Then, when loading the map again you encounter the Kidnapped Robot Problem (that is, if you don't know where to begin and with what orientation you have to find it yourself, and that's a pretty hard problem!

Try scan context/scan matching approaches if you don't know where to start) or, if known, you can set an initial pose and orientation. After you localized yourself in the initial pose it's a running the algorithm as if the pose and map were the ones you loaded in.

For confidentiality purposes I still cannot share the modifications done by my team but it's a pretty straightforward approach!

Hope this helps,
Good luck!

HappySamuel, 2021-07-05:
Hi @Huguet57 

Thanks for the sharing. Will try to download A-LOAM and see if i am able to do that according to your approach.

Anyone else have some thought or sharing on this kind of similar work?

Best,
Samuel

Huguet57, 2021-07-05:
@HappySamuel just bear in mind that ALOAM is not fit to use IMU data nor loop closures.

closed 2021-07-21 by stale[bot]

xiaoleicai, 2025-07-16:
Hi Samuel, did you succeed in using LIO-SAM for just localization?