TixiaoShan/LIO-SAM #268: Improving indoor performance

Jaetriel, 2021-07-15:
Hello guys, I've been using LIO-SAM with decent success outdoors, but when it comes to indoors I struggle to find the right params to run it with and get very mixed results, sometimes it's okay but most of the time the odometry is a lot more unstable. I've tried reducing the voxel leaf size as suggested, and I've also played around with the loop closure params and mapping interval but they don't seem to have huge impact. I am curious to find out what the reason is for the performance difference and what makes indoor mapping so much more difficult, but also what would be some ways to potentially improve the algorithm to also work well in indoor environments as it does outdoors. Thanks in advance!

Huguet57, 2021-07-18:
Hi Jaetriel,
What is your IMU rate? Me and my team were having trouble using this algorithm @55Hz, it became unstable very easily. Hopefully, we could increase to 200Hz and it all goes better now. Do you have videos of the unstabilization? Here are results of our indoor output:

[image]

[image]

[image]

Jaetriel, 2021-07-19:
@Huguet57 Hi, thanks for your answer. My IMU runs at 500hz raw data output, it can go up to 1KHz but I didn't see much improvement going from 500 to 1khz. My main issues right now are scanning smaller indoor rooms, it seems like the odometry isn't as good and starts becoming unstable and inconsistent, sometimes it drifts or tilts, sometimes even crashes if I rotate or move the sensors too quickly indoors. I will attach a few screenshots of maps we have, I might try to get a short video as well if I can find the time.
[image]
[image]
[image]

closed 2021-08-05 by stale[bot]

AHuguetUPC, 2021-09-09:
Hi both,
Have you checked changing the parameters at lines 62-65 of config/params.yaml? Those are what helped me the most going from outdoor to indoor.

We use a Vectornav VN-300.

Andreu

Jaetriel, 2021-09-10:
@AHuguetUPC hey, thanks for you suggestion! I have tried fiddling around with the voxel filter parameters, and in some cases I have seen noticeable improvement but I find that it's just kind of unstable overall when working indoor, sometimes I get decent results, sometimes LIO-SAM just crashes for me and I cannot really figure out a way to make it work consistently indoors, and maybe it is not really meant to, my knowledge doesn't go so far so I'm not sure

AHuguetUPC, 2021-09-10:
Using a branded LiDAR (Ouster, Velodyne, HESAI) I guess all the unstability of the algorithm comes from the IMU. When we were comparing our IMU output with the ones on the public datasets we noticed that our raw output was way noiser, have you tried applying a low pass filter beforehand?

We also found that LIO-SAM really outperformed others when outside going at 10m/s doing hairpin turns with our autonomous car. That's where the IMU really helped serving as an initial solution. Judging for your application, if you do not have a drone or anything similar that can go that fast, have you tried a loosely-coupled approach? Personally, we love the stability of A-LOAM that for localizing the robot with lower speeds (<10m/s) delivers outstanding results. We have only tried it outdoors, however.

Jaetriel, 2021-09-10:
@AHuguetUPC I've noticed that the public datasets are really stable, but I'm not sure how I could efficiently compare the output of the IMU used there versus the one we use, are there any tools that can quickly help with that?

Huguet57, 2021-09-10:
> @AHuguetUPC I've noticed that the public datasets are really stable, but I'm not sure how I could efficiently compare the output of the IMU used there versus the one we use, are there any tools that can quickly help with that?

I guess there's not much more you can do than comparing the output data (using ``rosrun rqt_plot rqt_plot``) of the two signals and how they behave on similar situations. Noisy IMU data with a slow moving vehicle on the inside can be a real pain.

Dhruv0208, 2022-09-28:
> Hi @Huguet57 , @Jaetriel
> 
> What kind of imu are both of you using? I am facing the same issue as Jaetriel, when come to indoor, the mapping result is similar to what Jaetriel show. I am using Ouster Lidar, therefore i use the built-in imu of Ouster lidar and pointcloud from ouster lidar to feed LIO-SAM. Ouster IMU gives 100Hz data. I also would like what parameters shall i set to get a good result in indoor.
> 
> Best, Samuel

Hi @HappySamuel I am also working on using Ouster internal IMu with LIO-SAM, can you guide me with it?
Thank you.