TixiaoShan/LIO-SAM #269: Files got deleted !!! - Strange Issue while saving the map

kjyothiswaroop, 2021-07-18:
Hello,

So I ran the LIO-SAM Algorithm on one of the recorded bag files. (campus_small_dataset.bag)
I wanted to save the map and so ran the rosservice command as given in the documentation except that instead of creating a separate LOAM folder I saved the map in my Downloads folder itself.

The command I ran :
rosservice rosservice call /lio_sam/save_map 0.2 "/Downloads".

After this, the map did get saved in my Downloads folder. But all the other files in my Downloads folder got deleted and I cant find them also in the Trash bin.

Is there any way to recover from this issue. Please provide help!!

Huguet57, 2021-07-18:
Hello,
Sorry to hear that. However, I think they can be recovered. Seems like you just ran this command (lines 359-361 mapOptimization.cpp):
`
// create directory and remove old files;`
`int unused = system((std::string("exec rm -r ") + saveMapDirectory).c_str()); `
`unused = system((std::string("mkdir -p ") + saveMapDirectory).c_str());
`

Since you put "Downloads" as your folder to save the map, it has removed old files (a.k.a. the "Downloads" folder). I would suggest you to look ways to recover files that have been removed by rm, the first I came up with: [https://unix.stackexchange.com/questions/10883/where-do-files-go-when-the-rm-command-is-issued](url).

I will suggest a pull request to change this methodology since it does not seem as an uncommon mistake to set a non-dedicated folder to save your maps.

kjyothiswaroop, 2021-07-18:
Sure. I will check for the files removed via rm. 
And I will create a pull request for the same. Thank you

closed 2021-08-05 by kjyothiswaroop

zhuzy-2018, 2022-08-17:
Fuck, you said change, until today has not changed ah ！！！！！ Get rid of the rm code ！！！！ Today, this line of code by the author deleted all the adaptations to LeGO-LOAM in my workspace, but not the LIO-SAM code. it's very good.

zhuzy-2018, 2022-08-17:
> Sure. I will check for the files removed via rm. And I will create a pull request for the same. Thank you

Fuck, you said change, until today has not changed ah ！！！！！ Get rid of the rm code ！！！！ Today, this line of code by the author deleted all the adaptations to LeGO-LOAM in my workspace, but not the LIO-SAM code. it's very good.

zhuzy-2018, 2022-08-17:
> Sure. I will check for the files removed via rm. And I will create a pull request for the same. Thank you

Fuck,  Get rid of the rm code ！！！！ Today, this line of code by the author deleted all the adaptations to LeGO-LOAM in my workspace, but not the LIO-SAM code. it's very good.

Huguet57, 2022-08-17:
Have you tried recovering them?

zhuzy-2018, 2022-08-17:
How

| |
zzy
|
|
***@***.***
|

---- Replied Message ----
| From | Andreu ***@***.***> |
| Date | 08/18/2022 00:06 |
| To | ***@***.***> |
| Cc | ***@***.******@***.***> |
| Subject | Re: [TixiaoShan/LIO-SAM] Files got deleted !!! - Strange Issue while saving the map (#269) |

Have you tried recovering them?

Huguet57, 2022-08-17:
> Hello, Sorry to hear that. However, I think they can be recovered. Seems like you just ran this command (lines 359-361 mapOptimization.cpp): ` // create directory and remove old files;` `int unused = system((std::string("exec rm -r ") + saveMapDirectory).c_str()); ` `unused = system((std::string("mkdir -p ") + saveMapDirectory).c_str()); `
> 
> Since you put "Downloads" as your folder to save the map, it has removed old files (a.k.a. the "Downloads" folder). I would suggest you to look ways to recover files that have been removed by rm, the first I came up with: https://unix.stackexchange.com/questions/10883/where-do-files-go-when-the-rm-command-is-issued.
> 
> I will suggest a pull request to change this methodology since it does not seem as an uncommon mistake to set a non-dedicated folder to save your maps.

In this comment I talked about it, see if this works. The pull request is commenting line 359...

zhuzy-2018, 2022-08-17:
Thanks any way. It's deep night now. I'll try it tomorrow.

| |
zzy
|
|
***@***.***
|

---- Replied Message ----
| From | Andreu ***@***.***> |
| Date | 08/18/2022 00:09 |
| To | ***@***.***> |
| Cc | ***@***.******@***.***> |
| Subject | Re: [TixiaoShan/LIO-SAM] Files got deleted !!! - Strange Issue while saving the map (#269) |

Hello, Sorry to hear that. However, I think they can be recovered. Seems like you just ran this command (lines 359-361 mapOptimization.cpp): // create directory and remove old files;int unused = system((std::string("exec rm -r ") + saveMapDirectory).c_str()); unused = system((std::string("mkdir -p ") + saveMapDirectory).c_str());

Since you put "Downloads" as your folder to save the map, it has removed old files (a.k.a. the "Downloads" folder). I would suggest you to look ways to recover files that have been removed by rm, the first I came up with: https://unix.stackexchange.com/questions/10883/where-do-files-go-when-the-rm-command-is-issued.

I will suggest a pull request to change this methodology since it does not seem as an uncommon mistake to set a non-dedicated folder to save your maps.

In this comment I talked about it, see if this works. The pull request is commenting line 359...