TixiaoShan/LIO-SAM #270: imu_preintegration and Map_optimization dies when using GTSAM 4.0.2

vicente879, 2021-07-26:
Hi, thank you for your work and resources!

I have been trying to make this package work with a gazebo Simulation using the Velodyne plugin and also with the real  sensor and a 9DOF IMU. 

I am using Ros Melodic i Ubuntu 18.04. I have followed the README steps for downloading and building the package. 
I have installed GSTAM 4.0.2 as required in the README. 

After everything is installed I get this error message even before I start playing my rosbag: the lio_sam_imuPreintegration and lio_sam_mapOptmization processes keep dying and restarting as seen in the error bellow my issue is with an Undefined Symbol, I have not seen this error in any of the other issues raised. 

`process[lio_sam_imuPreintegration-2]: started with pid [13505]
/home/main/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration: symbol lookup error: /home/main/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration: undefined symbol: _ZNK5gtsam16NoiseModelFactor5printERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFS6_mEE
process[lio_sam_imageProjection-3]: started with pid [13506]
process[lio_sam_featureExtraction-4]: started with pid [13507]
process[lio_sam_mapOptmization-5]: started with pid [13513]
process[lio_sam_rviz-6]: started with pid [13514]
[lio_sam_imuPreintegration-2] process has died [pid 13505, exit code 127, cmd /home/main/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/home/main/.ros/log/f3181d58-ee02-11eb-8a21-841b77053d6d/lio_sam_imuPreintegration-2.log].
log file: /home/main/.ros/log/f3181d58-ee02-11eb-8a21-841b77053d6d/lio_sam_imuPreintegration-2*.log
[lio_sam_imuPreintegration-2] restarting process
[ INFO] [1627298201.027332889]: ----> Image Projection Started.
process[lio_sam_imuPreintegration-2]: started with pid [13518]
/home/main/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration: symbol lookup error: /home/main/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration: undefined symbol: _ZNK5gtsam16NoiseModelFactor5printERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFS6_mEE
/home/main/catkin_ws/devel/lib/lio_sam/lio_sam_mapOptmization: symbol lookup error: /home/main/catkin_ws/devel/lib/lio_sam/lio_sam_mapOptmization: undefined symbol: _ZNK5gtsam16NoiseModelFactor5printERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFS6_mEE
[ INFO] [1627298201.103953467]: ----> Feature Extraction Started.
[lio_sam_imuPreintegration-2] process has died [pid 13518, exit code 127, cmd /home/main/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/home/main/.ros/log/f3181d58-ee02-11eb-8a21-841b77053d6d/lio_sam_imuPreintegration-2.log].
log file: /home/main/.ros/log/f3181d58-ee02-11eb-8a21-841b77053d6d/lio_sam_imuPreintegration-2*.log
[lio_sam_mapOptmization-5] process has died [pid 13513, exit code 127, cmd /home/main/catkin_ws/devel/lib/lio_sam/lio_sam_mapOptmization __name:=lio_sam_mapOptmization __log:=/home/main/.ros/log/f3181d58-ee02-11eb-8a21-841b77053d6d/lio_sam_mapOptmization-5.log].
log file: /home/main/.ros/log/f3181d58-ee02-11eb-8a21-841b77053d6d/lio_sam_mapOptmization-5*.log
[lio_sam_imuPreintegration-2] restarting process
process[lio_sam_imuPreintegration-2]: started with pid [13534]
[lio_sam_mapOptmization-5] restarting
 `

So I tried to reinstall GTSAM 4.0.2 and I noticed that when  GTSAM is not installed or when I install GTSAM 4.1.0 from Ubuntu PPA the Imu-Preintegration issue is resolved. Only the map-optimization issue remains (after playing the rosbag). 

`process[lio_sam_mapOptmization-5]: started with pid [5455]
[ INFO] [1627300996.909434086]: ----> Map Optimization Started.
[lio_sam_mapOptmization-5] process has died [pid 5455, exit code -11, cmd /home/main/catkin_ws/devel/lib/lio_sam/lio_sam_mapOptmization __name:=lio_sam_mapOptmization __log:=/home/main/.ros/log/5e2cb972-ee09-11eb-8a21-841b77053d6d/lio_sam_mapOptmization-5.log].
log file: /home/main/.ros/log/5e2cb972-ee09-11eb-8a21-841b77053d6d/lio_sam_mapOptmization-5*.log
[lio_sam_mapOptmization-5] restarting process
process[lio_sam_mapOptmization-5]: started with pid [5464]
[ INFO] [1627300997.410619265]: ----> Map Optimization Started.
`

From other issues in this repo I can see that the MapOptimization issue is because GTSAM 4.0.2 is not properly installed. Therefore this might be that GTSAM 4.0.2 is affecting the IMU pre-integration. In the error message just above you can see the output when _no GTSAM or GTSAM4.1.0_ is installed.

I'd appreciate if anyone could help me shed some light on this issue, specially in the undefined symbol part. Thank you

vicente879, 2021-08-03:
I have solved this issue. I realized that I had used catkin build instead of catkin_make to install the package. I have passed the package to a catkin_make workspace and this issue is solved.

closed 2021-08-03 by vicente879

mikolodz, 2024-04-17:
> I have solved this issue. I realized that I had used catkin build instead of catkin_make to install the package. I have passed the package to a catkin_make workspace and this issue is solved.

Hello, sorry to dig this out after almost 3 years, but what did you exactly mean by "I have passed the package to a catkin_make workspace and this issue is solved."? I'm asking because I face the very same problem with Isaac Sim. In my case, I built and installed the gtsam-4.0.2 and I'm building my lio-sam project in the catkin workspace (catkin_ws/src/ folder) using catkin_make. Still facing this issue.

Maybe you could help me out? Thanks!

REGATTE, 2025-01-10:
Any update on the ISAAC Sim? Im trying to build LIO_SAM with Isaac sim, but have issues with the robot moving on its own in the rviz viz, even though the robot is not moving. 

I figured out other issues such as point cloud ring field, by using another package.