TixiaoShan/LIO-SAM #271: Error on roslaunch

ManChrys, 2021-07-26:
Hello guys,

i just install the repo and when im running the launch file i have the following error.

[image]

[image]

gchbharath, 2021-07-28:
sudo apt-get install libparmetis-dev

ManChrys, 2021-07-28:
geo@geo:~$ sudo apt-get install libparmetis-dev
[sudo] password for geo: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libparmetis-dev is already the newest version (4.0.3-5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

ManChrys, 2021-07-30:
SOLUTION!

just delete every gtsam file ( i had 2 versions)

and make a clean install

gtsam (Georgia Tech Smoothing and Mapping library)

wget -O ~/Downloads/gtsam.zip https://github.com/borglab/gtsam/archive/4.0.2.zip
cd ~/Downloads/ && unzip gtsam.zip -d ~/Downloads/
cd ~/Downloads/gtsam-4.0.2/
mkdir build && cd build
cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
sudo make install -j8

closed 2021-07-30 by ManChrys