TixiaoShan/LIO-SAM #280: the map will be bad when the car turning

llxClover, 2021-08-28:
The imu data is in ENU(x-forword  y-left  z-up), so the two matrices are  identity matrix.
Once my car makes a turnning, the map will be overlap or flying.

why?

[image]

llxClover, 2021-08-28:
laser: velodyne16      IMU:  UM7

llxClover, 2021-08-28:
It may be tcaused by bad time sync between the lidar and IMU.
The first figure is the timeStamp for the west.bag which is provied by author.

The last
[image]
[image]
 figure  is the timeStamp for the Lee.bag which is recorded myself.
but, how to resolve the bad time sync?

llxClover, 2021-08-29:
I have resolved the trouble, but the method is offline.
I have sync the Lee.bag by python:

#!/usr/bin/env python2

import rospy
import rosbag
import sys
import datetime
import time

if sys.getdefaultencoding() != 'utf-8':
    reload(sys)
    sys.setdefaultencoding('utf-8')
bag_name = 'Lee.bag' 
out_bag_name = 'Lee_correct.bag' 
dst_dir = '/home/dt/Downloads/data/Lee/'

with rosbag.Bag(dst_dir+out_bag_name, 'w') as outbag:
    
    for topic, msg, t in rosbag.Bag(dst_dir+bag_name).read_messages():
        if topic =='/imu/data':
            stampFlag = msg.header.stamp
            outbag.write(topic, msg, msg.header.stamp)
        elif topic =='/imu/data_raw':
            outbag.write(topic, msg, msg.header.stamp)
        elif topic == '/velodyne_points':
            msg.header.stamp = stampFlag
            outbag.write(topic, msg, msg.header.stamp)

print("finished")
 
[image]

Jaetriel, 2021-09-03:
> It may be caused by bad time sync between the lidar and IMU.
> The figure is the timeStamp for the west.bag which is provied by author.
> [image]
> The figure is the timeStamp for the Lee.bag which is recorded myself.
> [image]
> but, how to resolve the bad time sync?

Hey, can I ask what tool you are using in the figure to check the time sync between IMU and lidar?

FishInWave, 2021-09-03:
> Hey, can I ask what tool you are using in the figure to check the time sync between IMU and lidar?

rosrun rqt_bag rqt_bag

llxClover, 2021-09-03:
rqt_bag    *.bag

llxClover, 2021-09-03:
> rosrun rqt_bag rqt_bag

rqt_bag *.bag

Jaetriel, 2021-09-03:
thank you, do you think this is an efficient way to time sync the data? I think it should be possible to achieve sensor synchronization real-time without having to rely on such a script, even though it seems to do a great job

llxClover, 2021-09-03:
I'm sorry！ I don't know how to do， but I have run LIO-SAM real-time by a ros-node successfully which is same to  the script.

FishInWave, 2021-09-03:
> I'm sorry！ I don't know how to do， but I have run LIO-SAM real-time by a ros-node successfully which is same to  the script.
> […](#)

Your IMU seems like running at 20Hz, which is so low. Usually, IMU runs at 100Hz or higher. I think maybe this is the reason why time-synchronization is so important for your bag.

llxClover, 2021-09-04:
> > I'm sorry！ I don't know how to do， but I have run LIO-SAM real-time by a ros-node successfully which is same to  the script.
> > […](#)
> 
> Your IMU seems like running at 20Hz, which is so low. Usually, IMU runs at 100Hz or higher. I think maybe this is the reason why time-synchronization is so important for your bag.

“ However the um7_driver has a default configuration which sets the transmission rate of all the parameters to 20 hz”。Haha， I used a default configuration. 
Thank you!

Jaetriel, 2021-09-06:
did this way of timesyncing improve the quality of your data drastically? it seems a bit brute force to me

llxClover, 2021-09-06:
> did this way of timesyncing a drastically? it seems a bit brute force to me

yes，it improves the quality of mapping

Jaetriel, 2021-09-06:
@llxClover thank you I think I might give it a try on my own data to see if I get better results

Huguet57, 2021-09-12:
Had the same problem with working with a IMU at 20Hz! Definitely was solved when going to 200Hz. @llxClover if you find that timesyncing improves the map that would be a huge discover. However in the README it specifically says that the IMU should work at a minimum of 200Hz rate.

wienans, 2021-10-04:
100 Hz works well too

closed 2022-03-02 by stale[bot]