TixiaoShan/LIO-SAM #282: How to incense density of global map

GigaFlopsis, 2021-09-08:
Thank you for a cool project. I want to use lio-sam for  indoor 3d mapping, however when I build a map it has a low density pcd model. All I found was reducing the size of the features to increase the density:
# voxel filter paprams
  odometrySurfLeafSize: 0.4                     # default: 0.4 - outdoor, 0.2 - indoor
  mappingCornerLeafSize: 0.2                    # default: 0.2 - outdoor, 0.1 - indoor
  mappingSurfLeafSize: 0.4                      # default: 0.4 - outdoor, 0.2 - indoor

when the value of c is too small, an error appears
```
[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.
```
 Maybe there is some more significant parameter for this?

mteastepp, 2021-11-16:
Hey GigaFlopsis, I am trying to use LIO-SAM for the same exact thing, 3D mapping. For me, I'm doing outdoor instead of indoor, but I ran into a similar issue with the Global map not being dense enough. I found the same thing that you did, where making the feature size smaller would increase density, but then throw those warnings. I was wondering if you did any further testing to improve the density? Here's what I've found to increase my density. 

Leave the feature size at the recommended settings. 

Change `surroundingkeyframeAddingDistThreshold` to be lower. I set mine to 0.25 (1/4 of the default)

Change `surroundingkeyframeAddingAngleThreshold` to be lower. I set mine to 0.05 (1/4 of the default). 

Change `globalMapVisualizationLeafSize` to be lower. I set mine to 0.5 (1/2 default). I'm not entirely sure if this impacts the saved PCD, but it seems to visually improve it

Change `globalMapVisualizationPoseDensity` to be higher. I set mine to 20 (2x default). Again, not sure if this actually impacts the saved PCD. Both this and the prior parameter did result in a larger amount of points in the PCD, but only marginally. So that may have just been due to some randomness. 

Finally, playing back your bag file at a slower speed seems to have helped quite a bit. You probably have to be careful with playing too slow though, as I believe this would impact the effective IMU rate. And as stated in the main Git page, an IMU with >200Hz is recommended. I have a 500Hz IMU and I did a playback at 0.75x speed. So an effective IMU rate of 375Hz. Unfortunately this won't work if you're relying specifically on the map produced in real-time. 

All of these changes took my density from 4.9 million points to 19.8 million points. 

Now these are all things that I found experimentally by just tweaking the parameters, playing my bag file, and seeing how it impacted the results. So I'm not actually sure how it impacts the underlying functionality, and these changes may actually result in something else breaking that I'm just not noticing. So feedback from someone more well-versed with the code would still be appreciated.

GigaFlopsis, 2021-11-18:
Everything turned out to be simpler. I just save RAW Pointcloud with localization by LION-SAM.

GigaFlopsis, 2021-11-18:
Yes. Looks the same)

farhad-dalirani, 2023-04-07:
@grischi @mteastepp Hello guys,
I want to create a denser global map, but I did not understand your method. Can you guide me with more detail?