TixiaoShan/LIO-SAM #283: Some questions about surface feature extraction

BLB0803, 2021-09-27:
Hello, Shan, I had read the article and watched the LIO-SAM and LVI-SAM demo on YouTube, there are amazing work.

And I'm reading the codes to get further understanding of whole system now.

BUT, I found something really confusing me, at

[featureExtraction.cpp](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/featureExtraction.cpp), function `extractFeatures()`,  **line 202,**

I saw that a point belongs to planar feature will be labeled as -1 in the array called `cloudLabel`.

In which part makes me confused is, at **line 227** of [featureExtraction.cpp](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/featureExtraction.cpp),  below the labeling part mentioned before, 

`if (cloudLabel[k] <= 0){
     surfaceCloudScan->push_back(extractedCloud->points[k]);
}`

It seems like you chosed not only points  labeled as -1 but also labeled as 0.

To check out this part, I visualized the  `surfaceCloud` that published by the publisher called `pubSurfacePoints` at rviz, here's the result:

[image]

Some points should not belongs to planar feature(like points of trees) had been extracted, cuz of pushing not only -1 but also 0 labeled points into surfaceCloudScan.

So I tryed to change 
`if (cloudLabel[k] <= 0){
     surfaceCloudScan->push_back(extractedCloud->points[k]);
}`
with
`if (cloudLabel[k] < 0){
     surfaceCloudScan->push_back(extractedCloud->points[k]);
}`

to save planar features properly, and the visualize result like below:
[image]

It seems like extracted some "real" planar features, although are sparse.

But after I changed that, **every time** when I test with the dataset named **campus_large_dataset.bag**, 

It crashed at the timestamp of`[PAUSED ]  Bag Time: 1572807852.632124   Duration: 430.792850 / 994.965750 `like below:

[image]

[image]

Although  I had double checked the `/config/params.yaml` to ensure parameters was configured properly.

And the WARN message is:

`[ WARN] [1632731906.350954159]: Large velocity, reset IMU-preintegration!`

Maybe it's because of lacking of features after I changed code like that, I guess.

Is this a bug or just I misunderstood the codes of that part?  Because I had checked the same part of LVI-SAM at [featureExtraction.cpp](https://github.com/TixiaoShan/LVI-SAM/blob/master/src/lidar_odometry/featureExtraction.cpp), it's exactly the same as:

`if (cloudLabel[k] <= 0){
     surfaceCloudScan->push_back(extractedCloud->points[k]);
}`

But in my opinion, a points labeled as '0' should not belongs to planar feature, since the value of each points in the label container `cloudLabel`, would be initialized as 0 by default.

So how could I fix this problem? Expecting for your reply.

Huguet57, 2021-10-03:
Hi,
I do not know the reasoning behind this part of the code but isn't true that if we take the edges of a cloud, everything else would belong in the surface category? I guess this is because it's easier to pinpoint which points belong to the edges (because they are normally singular discontinuities) that which ones belong to surfaces (because we do not have continuous resolution). So I guess classifying points that are definitely edges (1), that are definitely surfaces (-1) and the rest that might belong to surfaces (0) is what is being done here.

AlexReimann, 2021-12-13:
fyi:
From my experience the `degenerate` check can make issues if the amount of points is small:
https://github.com/TixiaoShan/LIO-SAM/blob/3a3d82fac4ba614ff2324be4ef83381b4256cc48/src/mapOptmization.cpp#L1225-L1247

When turning it off, the small campus dataset worked fine.

closed 2022-04-16 by stale[bot]