TixiaoShan/LIO-SAM #284: Question about map persistence

tomlogan501, 2021-09-30:
Hi,
Does the SLAM update the moving object during the map creation ?
I'm looking for a 3D SLAM with the same concept as hector_slam for map update, some erode concept for each linar observations.

Huguet57, 2021-10-03:
Hi!
If you are referring to moving objects leaving a trail, yes, in this SLAM they leave a trail. This SLAM is based on [LeGO-LOAM](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM)  (with the added 2-factor graph) with is based on [LOAM](https://github.com/HKUST-Aerial-Robotics/A-LOAM) (linked here the adaptation by the HKUST that simplifies, reorganizes and cleans up the original version), none of which have the ability to map moving objects.

I do not know if there's any broadly-used SLAM for that matter, I guess for cameras is a matter of optical flow but for LiDAR feels that it should be easier. Was that what you were asking?

tomlogan501, 2021-10-03:
It's exactly what I was asking. I thought 3D lidar slam got this feature but it seems not, indeed. I also try Kitware slam and it makes weird things of opposite moving object.

wienans, 2021-10-04:
@tomlogan501 there is SC-LIO-SAM which adds Scan Context to LIO-SAM but also outputs data to use with removert which can remove dynamic objects afterwards

closed 2022-03-02 by stale[bot]