TixiaoShan/LIO-SAM #285: The map is overlapping while looping, why?

XinrunChao, 2021-10-15:
lidar: lslidar-16(10Hz)
IMU: sanchi(100Hz)
While looping, the map is overlapping, how to correct it, or some tips for mapping?
Thank you!

[image]

zhouqigao, 2021-10-22:
hi,I also had the same problem before, but then I changed the starting point and walked around again,this overlapping phenomenon disappeared , which is very strange.do you have any solution now?

> lidar: lslidar-16(10Hz) IMU: sanchi(100Hz) While looping, the map is overlapping, how to correct it, or some tips for mapping? Thank you!
> 
> [image]

JACKLiuDay, 2022-02-14:
I have the same problem. When the test area is undulating terrain, the map will overlap at the looping point.

closed 2022-04-16 by stale[bot]