TixiaoShan/LIO-SAM #290: roslaunch Error on lio_sam_mapOptimization

MoFre01, 2021-11-10:
I want to use the LIO-Sam Repo to build a map from my recorded point clouds.
The build was successful and I am also able to run the launch file.
Unfortunately, the following error occurs during the execution of the launch file and the run of one of the example bagfiles (Park dataset):

[lio_sam_mapOptmization-4] restarting process
process[lio_sam_mapOptmization-4]: started with pid [46484]
[ INFO] [1636538223.480637248]: ----> Map Optimization Started.
[lio_sam_mapOptmization-4] process has died [pid 46484, exit code -11, cmd /home/User/CR_AAS/lioSam_ws/devel/lib/lio_sam/lio_sam_mapOptmization __name:=lio_sam_mapOptmization __log:=/home/User/.ros/log/aa0cce58-420b-11ec-8eb8-6b3f5bac804a/lio_sam_mapOptmization-4.log].
log file: /home/User/.ros/log/aa0cce58-420b-11ec-8eb8-6b3f5bac804a/lio_sam_mapOptmization-4*.log

At least RVIZ publishes some point clouds but I think, the trajectory seems to be wrong visualized, and also there is no map generated.

Could someone please help me?
Thank you very much in advance! :)

zpw6106, 2021-11-28:
@MoFre01  I have the same error, Don't you save it?

ZQR-1998, 2021-12-20:
I have the same error, Do you solved it?

SlammerD, 2021-12-23:
Have you modified anything in the code? Because mapOptimization takes the GPS data for optimization my first guess would be a deactivated GPS in params.yaml. When imu and lidar data does not come in right, usualy the imuPreintegration would show errors before.

closed 2022-04-16 by stale[bot]

TheSeanParker, 2024-04-25:
> @MoFre01 I have the same error, Do you solved it?

https://zhuanlan.zhihu.com/p/694553528,maybe help you ,although time has pasten 3 years

zhoupengwei, 2024-04-25:
@TheSeanParker This error related to the GTSAM library, you should build it without the MARCH_NATIVE supported as follows:
```bash
cmake -DCMAKE_BUILD_TYPE=Release \
           -DBUILD_SHARED_LIBS=ON \
           -DGTSAM_USE_QUATERNIONS=ON \
           -DGTSAM_USE_SYSTEM_EIGEN=ON \
           -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF \
           -DCMAKE_INSTALL_PREFIX=/usr/local ../
```
Then rebuild the LIO-SAM with it.