TixiaoShan/LIO-SAM #304: compatibility of ros version

kimth1757, 2022-01-07:
Hi, I'm Taehan. 
Always thanks to your contribution.
We are in the process of lidar map building for urban autonomous driving.
I have a few question about your algorithm.

1. Does your algorithm (LIO-SAM, LeGO-LOAM) work in ros noetic version?
    - I'v already use your LeGO-LOAM algorithm in ros kinetic version. 
2. In sparse area such as large intersection, do you think is your algorithm work well?
3. Do you have experience with lidar + RTK based map building?

Actually, I'm having a hard time build a cm-level accuracy lidar map of the urban area.

Thanks in advance.
Taehan.

dsb6063, 2022-01-07:
Follow suggestions in # 206 and  algorithm with run in noetic.

catkin_make in ROS Noetic [Error] #206

dsb6063, 2022-01-07:
2. In sparse area such as large intersection, do you think is your algorithm work well? I have only observed failure indoors because edges were to close and another time with a backpack setup over rough terrain in a clearing that was flat.  It depends on the quality of scanner 16 vs 64 and high quality IMU is a must for cm level.
3. Do you have experience with lidar + RTK based map building?

The algorithm uses the gps elevation to constrain the z-axis. The algorithm uses horizontal in the static transform for a known place. It is difficult to get the data to converge on xyz at cm level without post processing all the data. For online slam, you can do RTK with corrections to help with the z.   I am currently working on these cm level items.

kimth1757, 2022-01-07:
> 2. In sparse area such as large intersection, do you think is your algorithm work well? I have only observed failure indoors because edges were to close and another time with a backpack setup over [...]

Many thanks to your kind reply!
Taehan.

kimth1757, 2022-01-18:
Thanks for your comment! @dsb6063

closed 2022-01-18 by kimth1757