TixiaoShan/LIO-SAM #309: Point cloud ring channel not availiable

DaddyWesker, 2022-02-01:
Hello and thanks for your code.

I've been trying to run your code on the kitti rosbag (but not one in your google drive). So, I've created rosbag using kitti2bag and tried to play it using command 
```
rosbag play --clock kitti_2011_09_29_drive_0071_synced.bag -r 3 /kitti/oxts/gps/fix:=/gps/fix /kitti/oxts/gps/vel:=/gps/vel /kitti/oxts/imu:=/imu_correct /kitti/oxts/imu:=/imu_raw /kitti/velo/pointcloud:=/points_raw
```
Unfortunately, I'm getting this `[ERROR] [1643712018.057128984]: Point cloud ring channel not available, please configure your point cloud data!` in the terminal, where i've launched lio_sam. I guess there should be some sort of pre-processing for the kitti data or maybe post-processing for the rosbag to add this channel. But where to find this pre/postprocessing? 

Thanks in advance.

yijing2050, 2022-10-22:
> add this channel.

Hello，have you solved the problem?

DaddyWesker, 2022-10-22:
> > add this channel.
> 
> Hello，have you solved the problem?

Hello, no.

Raymond888888, 2022-10-23:
Hello, I have the same problem.

yijing2050, 2022-10-26:
> Hello, I have the same problem.

hello,目前我找到的有人解决的方案。固态激光雷达尚未经过 LIO-SAM 的广泛测试。这里也使用外部 IMU，而不是内部 IMU。对此类激光雷达的支持基于对机械激光雷达代码库的最小更改。需要使用定制的 livox_ros_driver 来发布 LIO-SAM 可以处理的点云格式。https://blog.csdn.net/BIT_HXZ/article/details/127210765

Raymond888888, 2022-10-26:
> > Hello, I have the same problem.
> 
> hello,目前我找到的有人解决的方案。固态激光雷达尚未经过 LIO-SAM 的广泛测试。这里也使用外部 IMU，而不是内部 IMU。对此类激光雷达的支持基于对机械激光雷达代码库的最小更改。需要使用定制的 livox_ros_driver 来发布 LIO-SAM 可以处理的点云格式。https://blog.csdn.net/BIT_HXZ/article/details/127210765

谢谢！！

ESXNSI, 2023-05-05:
I solved this problem.

PanterSoft, 2023-09-19:
How did you solve this problem @ESXNSI

mikolodz, 2024-04-04:
> I solved this problem.

@ESXNSI Hello, can you please tell us how did you solve it and what was the issue? I'm working on setting up the IsaacSim as an input for the LIO-SAM. I'm posting the ouster 128, 1024 10Hz 3d lidar output in the /os_cloud_node/points topic. It's not a solid-state lidar, so I don't see any reason why would I use the custom livox_ros_driver for that?

iampaulidrobo, 2024-10-07:
Hi,I encounter the same error with ubuntu 24 noble and ros2 jazzy.
I went to the problem statement code [line](https://github.com/TixiaoShan/LIO-SAM/blob/0be1fbe6275fb8366d5b800af4fc8c76a885c869/src/imageProjection.cpp#L260)
and changed it to:

>  if (ringFlag == -1)
>             {
>                 if (sensor == SensorType::VELODYNE) {
>                     ringFlag = 2;
>                 } else {
>                     RCLCPP_ERROR(get_logger(), "Point cloud ring channel not available, please configure your point cloud data!");
>                     rclcpp::shutdown();
>                 }
>             }

 Building and running again  worked for me.
 Hope it helps someone else too.