TixiaoShan/LIO-SAM #313: [ INFO] [1645083917.209370774]: ----> Feature Extraction Started. double free or corruption (out) [lio_sam_featureExtraction-4] process has died

hichamhendy, 2022-02-17:
I am facing a problem with launching LioSam. It worked just fine before. However it stopped working due to double free or corruption (out)
Here is a snap from log
`[roslaunch][ERROR] 2022-02-17 08:45:11,141: [lio_sam_featureExtraction-4] process has died [pid 3694, exit code -6, cmd /home/rnd/catkin_ws/devel/lib/lio_sam/lio_sam_featureExtraction __name:=lio_sam_featureExtraction __log:=/home/rnd/.ros/log/7c99859e-8fc5-11ec-bde8-709cd1531f15/lio_sam_featureExtraction-4.log].
log file: /home/rnd/.ros/log/7c99859e-8fc5-11ec-bde8-709cd1531f15/lio_sam_featureExtraction-4*.log
[roslaunch][INFO] 2022-02-17 08:45:11,141: [lio_sam_featureExtraction-4] restarting process
[roslaunch][INFO] 2022-02-17 08:45:11,141: process[lio_sam_featureExtraction-4]: restarting os process
[roslaunch][INFO] 2022-02-17 08:45:11,141: process[lio_sam_featureExtraction-4]: start w/ args [[u'/home/rnd/catkin_ws/devel/lib/lio_sam/lio_sam_featureExtraction', u'__name:=lio_sam_featureExtraction', u'__log:=/home/rnd/.ros/log/7c99859e-8fc5-11ec-bde8-709cd1531f15/lio_sam_featureExtraction-4.log']]`

arjunskumar, 2022-02-21:
@hichamhendy Have you fixed the issue?

hichamhendy, 2022-02-21:
Ja. I reinstalled gtsam and built the workspace again

arjunskumar, 2022-02-22:
@hichamhendy which version of GTSAM you've installed ?

hichamhendy, 2022-02-23:
Follow that here
`wget -O ~/Downloads/gtsam.zip https://github.com/borglab/gtsam/archive/4.0.0-alpha2.zip
cd ~/Downloads/ && unzip gtsam.zip -d ~/Downloads/
cd ~/Downloads/gtsam-4.0.0-alpha2/
mkdir build && cd build
cmake ..
sudo make install`

and catkin build clean -y in your workspace

closed 2022-02-23 by hichamhendy