TixiaoShan/LIO-SAM #314: won't compile - no member named ‘serialize’

MGforest, 2022-02-22:
Hi, 
I am trying to build the package with catkin_make. 
The error is:
```
/usr/include/flann/util/serialization.h:34:14: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
   34 |         type.serialize(ar);
      |         ~~~~~^~~~~~~~~

```
I am using Ubuntu 20.04, ROS noetic and  installed all dependencies. Nothing I found in forums helped. Maybe you have an idea. Many thanks in advance!

Full log: 

```
[ 90%] Building CXX object LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/src/mapOptmization.cpp.o
[ 96%] Built target lio_sam_imageProjection
/home/<user>/catkin_ws/src/LIO-SAM/src/mapOptmization.cpp: In member function ‘bool mapOptimization::saveMapService(lio_sam::save_mapRequest&, lio_sam::save_mapResponse&)’:
/home/<user>/catkin_ws/src/LIO-SAM/src/mapOptmization.cpp:365:11: warning: variable ‘unused’ set but not used [-Wunused-but-set-variable]
  365 |       int unused = system((std::string("exec rm -r ") + saveMapDirectory).c_str());
      |           ^~~~~~
/home/<user>/catkin_ws/src/LIO-SAM/src/mapOptmization.cpp: In member function ‘void mapOptimization::publishFrames()’:
/home/<user>/catkin_ws/src/LIO-SAM/src/mapOptmization.cpp:1741:37: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
 1741 |             if (lastSLAMInfoPubSize != cloudKeyPoses6D->size())
      |                 ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/flann/util/matrix.h:35,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.10/pcl/kdtree/impl/kdtree_flann.hpp:44,
                 from /usr/include/pcl-1.10/pcl/kdtree/kdtree_flann.h:236,
                 from /home/<user>/catkin_ws/src/LIO-SAM/include/utility.h:24,
                 from /home/<user>/catkin_ws/src/LIO-SAM/src/mapOptmization.cpp:1:
/usr/include/flann/util/serialization.h: In instantiation of ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’:
/usr/include/flann/util/serialization.h:42:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:51:33:   required from ‘static void flann::serialization::Serializer<T>::load(InputArchive&, T&) [with InputArchive = flann::serialization::LoadArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:298:28:   required from ‘Archive& flann::serialization::InputArchive<Archive>::operator&(T&) [with T = std::unordered_map<unsigned int, std::vector<unsigned int> >; Archive = flann::serialization::LoadArchive]’
/usr/include/flann/util/lsh_table.h:319:10:   required from ‘void flann::lsh::LshTable<ElementType>::serialize(Archive&) [with Archive = flann::serialization::LoadArchive; ElementType = float]’
/usr/include/flann/util/serialization.h:34:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = flann::lsh::LshTable<float>]’
/usr/include/flann/util/serialization.h:42:22:   [ skipping 6 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/flann/util/serialization.h:34:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:42:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::LoadArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:51:33:   required from ‘static void flann::serialization::Serializer<T>::load(InputArchive&, T&) [with InputArchive = flann::serialization::LoadArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:298:28:   required from ‘Archive& flann::serialization::InputArchive<Archive>::operator&(T&) [with T = flann::LshIndex<flann::L2_Simple<float> >; Archive = flann::serialization::LoadArchive]’
/usr/include/flann/algorithms/lsh_index.h:204:9:   required from ‘void flann::LshIndex<Distance>::loadIndex(FILE*) [with Distance = flann::L2_Simple<float>; FILE = _IO_FILE]’
/usr/include/flann/algorithms/lsh_index.h:201:10:   required from here
/usr/include/flann/util/serialization.h:34:14: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
   34 |         type.serialize(ar);
      |         ~~~~~^~~~~~~~~
/usr/include/flann/util/serialization.h: In instantiation of ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’:
/usr/include/flann/util/serialization.h:42:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:56:33:   required from ‘static void flann::serialization::Serializer<T>::save(OutputArchive&, const T&) [with OutputArchive = flann::serialization::SaveArchive; T = std::unordered_map<unsigned int, std::vector<unsigned int> >]’
/usr/include/flann/util/serialization.h:316:28:   required from ‘Archive& flann::serialization::OutputArchive<Archive>::operator&(const T&) [with T = std::unordered_map<unsigned int, std::vector<unsigned int> >; Archive = flann::serialization::SaveArchive]’
/usr/include/flann/util/lsh_table.h:319:10:   required from ‘void flann::lsh::LshTable<ElementType>::serialize(Archive&) [with Archive = flann::serialization::SaveArchive; ElementType = float]’
/usr/include/flann/util/serialization.h:34:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = flann::lsh::LshTable<float>]’
/usr/include/flann/util/serialization.h:42:22:   [ skipping 6 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/flann/util/serialization.h:34:9:   required from ‘static void flann::serialization::access::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:42:22:   required from ‘void flann::serialization::serialize(Archive&, T&) [with Archive = flann::serialization::SaveArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:56:33:   required from ‘static void flann::serialization::Serializer<T>::save(OutputArchive&, const T&) [with OutputArchive = flann::serialization::SaveArchive; T = flann::LshIndex<flann::L2_Simple<float> >]’
/usr/include/flann/util/serialization.h:316:28:   required from ‘Archive& flann::serialization::OutputArchive<Archive>::operator&(const T&) [with T = flann::LshIndex<flann::L2_Simple<float> >; Archive = flann::serialization::SaveArchive]’
/usr/include/flann/algorithms/lsh_index.h:198:9:   required from ‘void flann::LshIndex<Distance>::saveIndex(FILE*) [with Distance = flann::L2_Simple<float>; FILE = _IO_FILE]’
/usr/include/flann/algorithms/lsh_index.h:195:10:   required from here
/usr/include/flann/util/serialization.h:34:14: error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
make[2]: *** [LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/build.make:63: LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/src/mapOptmization.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3007: LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j48 -l48" failed
```

dagata-mining, 2022-02-26:
Any updates on that?

MGforest, 2022-02-28:
Hi, 

I thought I did the trick with rearanging the include before, which did not work. I realised on friday, that I did it wrong. 

I now did it correctly and the package compiled. 
Here the corrected file catkin_ws/src/LIO-SAM/include/utility.h:

```
#pragma once
#ifndef _UTILITY_LIDAR_ODOMETRY_H_
#define _UTILITY_LIDAR_ODOMETRY_H_
#define PCL_NO_PRECOMPILE 

#include <ros/ros.h>

#include <std_msgs/Header.h>
#include <std_msgs/Float64MultiArray.h>
#include <sensor_msgs/Imu.h>
#include <sensor_msgs/PointCloud2.h>
#include <sensor_msgs/NavSatFix.h>
#include <nav_msgs/Odometry.h>
#include <nav_msgs/Path.h>
#include <visualization_msgs/Marker.h>
#include <visualization_msgs/MarkerArray.h>

#include <pcl/point_cloud.h>
#include <pcl/point_types.h>
#include <pcl/search/impl/search.hpp>
#include <pcl/range_image/range_image.h>
#include <pcl/kdtree/kdtree_flann.h>
#include <pcl/common/common.h>
#include <pcl/common/transforms.h>
#include <pcl/registration/icp.h>
#include <pcl/io/pcd_io.h>
#include <pcl/filters/filter.h>
#include <pcl/filters/voxel_grid.h>
#include <pcl/filters/crop_box.h> 
#include <pcl_conversions/pcl_conversions.h>

#include <tf/LinearMath/Quaternion.h>
#include <tf/transform_listener.h>
#include <tf/transform_datatypes.h>
#include <tf/transform_broadcaster.h>

#include <opencv2/opencv.hpp>

#include <vector>
#include <cmath>
#include <algorithm>
#include <queue>
#include <deque>
#include <iostream>
#include <fstream>
#include <ctime>
#include <cfloat>
#include <iterator>
#include <sstream>
#include <string>
#include <limits>
#include <iomanip>
#include <array>
#include <thread>
#include <mutex>
```

closed 2022-02-28 by MGforest