TixiaoShan/LIO-SAM #318: error after bulding enviroment - catkin_make

macnack, 2022-03-14:
Im using Ubuntu 18.04.6 LTS. Melodic ROS, acording to readme after catkin_make, i get answer: 
`Base path: /home/macko-pc/catkin_ws
Source space: /home/macko-pc/catkin_ws/src
Build space: /home/macko-pc/catkin_ws/build
Devel space: /home/macko-pc/catkin_ws/devel
Install space: /home/macko-pc/catkin_ws/install
Creating symlink "/home/macko-pc/catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/macko-pc/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/macko-pc/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/macko-pc/catkin_ws/install -G Unix Makefiles" in "/home/macko-pc/catkin_ws/build"
####
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/macko-pc/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/macko-pc/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - lio_sam
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'lio_sam'
-- ==> add_subdirectory(LIO-SAM)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.4
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
[... 287 lines cut ...]
                 from /home/macko-pc/catkin_ws/src/LIO-SAM/include/utility.h:27,
                 from /home/macko-pc/catkin_ws/src/LIO-SAM/src/mapOptmization.cpp:1:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
[ 95%] Linking CXX executable /home/macko-pc/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration
/usr/bin/ld: cannot find -lBoost::timer
collect2: error: ld returned 1 exit status
LIO-SAM/CMakeFiles/lio_sam_imuPreintegration.dir/build.make:759: recipe for target '/home/macko-pc/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration' failed
make[2]: *** [/home/macko-pc/catkin_ws/devel/lib/lio_sam/lio_sam_imuPreintegration] Error 1
CMakeFiles/Makefile2:3207: recipe for target 'LIO-SAM/CMakeFiles/lio_sam_imuPreintegration.dir/all' failed
make[1]: *** [LIO-SAM/CMakeFiles/lio_sam_imuPreintegration.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable /home/macko-pc/catkin_ws/devel/lib/lio_sam/lio_sam_mapOptmization
/usr/bin/ld: cannot find -lBoost::timer
collect2: error: ld returned 1 exit status
LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/build.make:759: recipe for target '/home/macko-pc/catkin_ws/devel/lib/lio_sam/lio_sam_mapOptmization' failed
make[2]: *** [/home/macko-pc/catkin_ws/devel/lib/lio_sam/lio_sam_mapOptmization] Error 1
CMakeFiles/Makefile2:2868: recipe for target 'LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/all' failed
make[1]: *** [LIO-SAM/CMakeFiles/lio_sam_mapOptmization.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

Today20180409, 2022-03-30:
I have the same problem, did you solve it？ @macnack

THLL245, 2022-04-06:
I too am having this same problem, any solutions?

Today20180409, 2022-04-07:
> 
I solved this problem by following two methods：
1.Please refer to this link [link](https://github.com/TixiaoShan/LIO-SAM/issues/229)；
2.I downgraded the version of gtsam from 4.0.3 to 4.0.2，then it compiles fine. This means it's a gtsam version problem。
The second method is recommended，the first method compiles normally, but the node will exit abnormally at runtime.
hope this helps.

aladin2022, 2022-04-11:
Hi , can you help me with the directory or a link to install GTSAM 4.0.2

closed 2022-04-28 by stale[bot]