TixiaoShan/LIO-SAM #319: cannot launch node of type [lio_sam/lio_sam_imuPreintegration]: Cannot locate node of type [lio_sam_imuPreintegration] in package [lio_sam].

Xu-Huitong, 2022-03-22:
When I run the command "roslaunch lio_sam run.launch", I get the following error, while rviz can be opened.
ERROR: cannot launch node of type [lio_sam/lio_sam_imuPreintegration]: Cannot locate node of type [lio_sam_imuPreintegration] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [lio_sam/lio_sam_imageProjection]: Cannot locate node of type [lio_sam_imageProjection] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [lio_sam/lio_sam_featureExtraction]: Cannot locate node of type [lio_sam_featureExtraction] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [lio_sam/lio_sam_mapOptmization]: Cannot locate node of type [lio_sam_mapOptmization] in package [lio_sam]. Make sure file exists in package path and permission is set to executable (chmod +x)

houcemsaidani, 2022-03-25:
same... waiting for a solution

houcemsaidani, 2022-03-25:
Hey i found this and it worked like a charm for me.
The solution is to specify the package to compile
just type:
catkin_make -DCATKIN_WHITELIST_PACKAGES="lio_sam"

i found the solution here:
https://zhuanlan-zhihu-com.translate.goog/p/419116270?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=ar

Xu-Huitong, 2022-03-29:
> Hey i found this and it worked like a charm for me. The solution is to specify the package to compile just type: catkin_make -DCATKIN_WHITELIST_PACKAGES="lio_sam"
> 
> i found the solution here: https://zhuanlan-zhihu-com.translate.goog/p/419116270?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=ar

thanks

closed 2022-04-18 by stale[bot]

nyxrobotics, 2022-12-12:
I have the same problem.
I am using Ubuntu 18.04+ROS melodic.
This bug should not be closed.