TixiaoShan/LIO-SAM #321: how to solve the problem "what():  PreintegratedImuMeasurements::integrateMeasurement: dt <=0"?

nonlinear1, 2022-03-24:
I have the same errors as the following link "https://github.com/TixiaoShan/LIO-SAM/issues/247",that is terminate called after throwing an instance of  std::runtime_error'
  what():  PreintegratedImuMeasurements::integrateMeasurement: dt <=0
I have try the methods which have given by "https://github.com/TixiaoShan/LIO-SAM/issues/247". But the errors remain. My cofiguration as following:
os:ubuntu20.04, gtsam 4.02
imu: 3DM-gx5-25
lidar: Pandar xt16
Could someone know how to solve this errors?

closed 2022-04-18 by stale[bot]

hjl66662, 2022-07-28:
I have the same problem as you. Have you solved it

ycl-web, 2022-11-29:
Please tell me if you have solved this problem, I also encountered this problem.thank you very much

OTTFFIVE, 2023-03-29:
The timestamp of the imu data has regressed, and it is necessary to filter the incorrect data

LordonCN, 2024-02-03:
> The timestamp of the imu data has regressed, and it is necessary to filter the incorrect data

At the same time, incorrect data which dt > 0.2 should also be filtered.