TixiaoShan/LIO-SAM #326: Not enough features! Only 0 edge and 0 planar features available.

JokerJohn, 2022-03-31:
I try to adjust LIO_SAM system with 6-axis imu before,  such as stim300, it works very well.  When I use the senor combination of Pandar xT32 and BMI085(6-axis MEMS), I faced many problems. As the below picture shows, **Not enough features! Only 0 edge and 0 planar features available**. I can guarantee that the extrinsic parameters from lidar to imu are correct. 

[image]

[image]

my params.yaml:
`lio_sam:

  # Topics
  pointCloudTopic: "/hesai/pandar"               # Point cloud data
  imuTopic: "/alphasense/imu"

  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "gps_odom"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useGps: false
  useImuHeadingInitialization: false           # if using GPS data, set to "true"
  useGpsElevation: false                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 0.02                      # m^2, threshold for using GPS data

  # Export settings
  savePCD: true                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/home/xchu/ramlab_ws/src/LIO-SAM/data/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation
  scene_name: "map"

  #   Sensor Settings
  sensor: hesai                            # lidar sensor type, 'velodyne' or 'ouster' or 'livox'
  N_SCAN: 32                                  # number of lidar channel (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
  Horizon_SCAN: 2000                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
  downsampleRate: 1                           # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1
  lidarMinRange: 0.5                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 100.0                       # default: 1000.0, maximum lidar range to be used

  #  sensor: velodyne                            # lidar sensor type, 'velodyne' or 'ouster' or 'livox'
  #  N_SCAN: 16                                  # number of lidar channel (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
  #  Horizon_SCAN: 1800                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
  #  downsampleRate: 1                           # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1
  #  lidarMinRange: 1.0                          # default: 1.0, minimum lidar range to be used
  #  lidarMaxRange: 1000.0                       # default: 1000.0, maximum lidar range to be used

  # IMU Settings
  imuFrequence: 400
  imuAccNoise: 0.0011501915187049582
  imuGyrNoise: 5.084312924828687e-05
  imuAccBiasN: 0.06080652138668933
#  imuAccBias_N: [ 0.06080652138668933, 0.08353074835853214, 0.057072968234636895 ]
  imuGyrBiasN: -0.0015351229643790084
#  imuGyrBias_N: [ -0.0015351229643790084, -0.0013449146576507546, 0.00030127855524786183 ]
  imuGravity: 9.804989173462031
#  imuGravity_N: [ 0.01165152782783894, 9.804989173462031, -0.008749296634685332 ]
  imuRPYWeight: 0.01

  #
  #  # Extrinsics (lidar -> IMU)
  extrinsicTrans: [ -0.001, -0.00855, 0.055 ]
    #  extrinsicRot: [ -0.0000000, -1.0000000,  0.0000000,
    #                  1.0000000, -0.0000000,  0.0000000,
    #                  0.0000000,  0.0000000, -1.0000000 ]
    #  extrinsicRPY: [ -0.0000000, -1.0000000,  0.0000000,
    #                  1.0000000, -0.0000000,  0.0000000,
    #                  0.0000000,  0.0000000, -1.0000000 ]
[... 147 lines cut ...]
roll: 3.141593, pitch: -0.000000, yaw: -1.570796

end time from pcd and size: 0.100178003311, 58084
IMU acc: 
x: 0.172266, y: -0.299790, z: 9.814834
IMU gyro: 
x: 0.102162, y: 0.004368, z: 0.000746
IMU roll pitch yaw: 
roll: 3.141593, pitch: -0.000000, yaw: -1.570796

[ WARN] [1648758938.083301867]: Not enough features! Only 0 edge and 0 planar features available.
IMU acc: 
x: 0.141521, y: -0.320127, z: 9.809570
IMU gyro: 
x: 0.112602, y: 0.000959, z: 0.004155
IMU roll pitch yaw: 
roll: 3.141593, pitch: -0.000000, yaw: -1.570796

IMU acc: 
x: 0.118193, y: -0.338071, z: 9.790908
IMU gyro: 
x: 0.114626, y: -0.001065, z: 0.005966
IMU roll pitch yaw: 
roll: 3.141593, pitch: -0.000000, yaw: -1.570796
`

closed 2022-04-01 by JokerJohn

oskarmue, 2022-07-01:
Hey, how did you fix that issue?

JasonWang9805, 2022-09-16:
why you use 6-axis,but output rpy?

harisbinyousaf24, 2023-07-04:
@JokerJohn Can you tell me how did you resolve that issue. I am currently working on LIO-SAM and i am facing the same issue. Your help would be appreciated.

yuma116, 2023-07-23:
Also, I want to information or code for fix. Could you share it?