TixiaoShan/LIO-SAM #331: Unknown frame map

DevAB-Git, 2022-04-11:
Hi,

I'm new to robotics. I'm trying to use this package with ROS Noetic (roslaunch lio_sam run.launch). However, my rviz shows that there is no transformation from base_link (chassis_link, imu_link, navast_link, and velodyne_link) to frame map. The base_link status is fixed frame does not exist and the Global Status is unknown frame map.

I think it is a very basic problem. I have read all the issues but could not find a similar problem faced by anyone else :(

Thanks in advanc
[image]
e

aladin2022, 2022-04-11:
I am  facing the same problem, any solution ?

TixiaoShan, 2022-04-11:
Was the package built successfully? 

There are a few changes that need to be done for Noetic.

https://github.com/TixiaoShan/LIO-SAM/issues/206#issuecomment-1095370894

DevAB-Git, 2022-04-11:
Morning Tixiao Shan,
Thank you for your reply. Yes, the package was built successfully. I'm using the noetic version from 
https://github.com/juliangaal/LIO-SAM

Note: I don't have a robot. I'm using ROS and Gazebo. I tried to run roscore first and then roslaunch lio_sam run.launch but have the same error. 

Please let me know if any other information is required. Thanks in advance.

TixiaoShan, 2022-04-12:
I don't have experience running LIO-SAM with Gazebo. But some users have successfully done it before.

https://github.com/engcang/SLAM-application

DevAB-Git, 2022-04-19:
Thanks Tixiao Shan for all the support and suggestions. I managed to run it :)
@[aladin2022](https://github.com/aladin2022) first run a gazebo package that can launch your robot (urdf) and then run the lio_sam package. It is noted that you have to adjust the parameters according to your robot (urdf).

closed 2022-04-19 by DevAB-Git

LiuYiKai6, 2022-12-02:
hello, may I ask how you solved the problem?

LiuYiKai6, 2022-12-02:
> Thanks Tixiao Shan for all the support and suggestions. I managed to run it :) @[aladin2022](https://github.com/aladin2022) first run a gazebo package that can launch your robot (urdf) and then run [...]

hello, may I ask how you solved the problem?

DeepDuke, 2023-01-30:
I met this problem in ros noetic. No frame called map. Could u tell me how do u fix it? Thank you!

SEUZTh, 2023-07-05:
> I met this problem in ros noetic. No frame called map. Could u tell me how do u fix it? Thank you!我在 ros noetic 中遇到了这个问题。没有称为地图的框架。你能告诉我你怎么解决它吗？谢谢！

Hi, have you solved the problem?

mnoweo, 2023-08-19:
Modify Fixed Frame in Global Options and map to base_link. 
Different datasets modify parameters in config/params.yaml. 
For example, Rotation dataset's imu topic needs to be changed to imuTopic: "imu_correct", and "extrinsicRot" and "extrinsicRPY" need to be set to rotation matrix.

mnoweo, 2023-08-19:
> 

Modify Fixed Frame in Global Options and map to base_link. 
Different datasets modify parameters in config/params.yaml. 
For example, Rotation dataset's imu topic needs to be changed to imuTopic: "imu_correct", and "extrinsicRot" and "extrinsicRPY" need to be set to rotation matrix.