TixiaoShan/LIO-SAM #341: odometry/gps not published

kimth1757, 2022-05-04:
Dear @TixiaoShan ,
Thanks for sharing your project.

I have successfully run your code and was able to get a lidar map. Currently, I'm using lidar point cloud data (Hesai Pandar64) and imu data (pwrpak7D + KVH1750). But I'm not using gps data yet. So, I'm not running module_navsat.launch yet.
Now, I would like to integrate gps data by running module_navsat.launch.

`<launch>

    <arg name="project" default="lio_sam"/>

    <env name="ROSCONSOLE_CONFIG_FILE" value="$(find lio_sam)/launch/include/rosconsole/rosconsole_error.conf"/>
    
    
    <node pkg="robot_localization" type="ekf_localization_node" name="ekf_gps" respawn="true">
        <remap from="odometry/filtered" to="odometry/navsat" />
    </node>

    
    <node pkg="robot_localization" type="navsat_transform_node" name="navsat" respawn="true">
        
        <remap from="/imu/data" to="/imu_correct" />
        <remap from="/gps/fix" to="/gps/fix" />
        <remap from="/odometry/filtered" to="/odometry/navsat" />
    </node>

</launch>`

"imu_correct" is manually transformed by myself and it is alined with REB105 convention.
"gps/fix" is available.
If I launch the module_navsat.launch files, I cannot get /odometry/gps and /odometry/navsat. I think it seems that navsat and ekf_gps nodes are not working well.

I want you to give me some advice.
Thanks,
Taehan.

learnhard22, 2022-07-26:
I have the same problem. Have you already been able to solve it?

Kind regards,
Lennard

pauldeee, 2022-07-26:
When using my own custom data, GPS did not function for me until I added it to the TF tree. For example my GPS frame name is gnss1, so I changed [this line](https://github.com/TixiaoShan/LIO-SAM/blob/9991daa47eb76124659570daab299bf877dde8bb/launch/include/config/robot.urdf.xacro#L31) to my GPS frame.

HomieRegina, 2022-09-18:
Hi, I have the same problem. Have you figured out how to remap the topics here?
Looking forward to your reply.

xingyuuchen, 2022-11-25:
> I have the same problem. Have you already been able to solve it?
> 
> Kind regards, Lennard

Probably you did not install the `robot-localization` package.

andrewangbl, 2024-06-17:
I think it is rosbag problem. Try this python code to generate kitti rosbag, it solves my problem.#!env python
# -*- coding: utf-8 -*-

import sys

try:
    import pykitti
except ImportError as e:
    print('Could not load module \'pykitti\'. Please run `pip install pykitti`')
    sys.exit(1)

import tf
import os
import cv2
import rospy
import rosbag
from tqdm import tqdm
from tf2_msgs.msg import TFMessage
from datetime import datetime
from std_msgs.msg import Header
from sensor_msgs.msg import CameraInfo, Imu, PointField, NavSatFix
import sensor_msgs.point_cloud2 as pcl2
from geometry_msgs.msg import TransformStamped, TwistStamped, Transform
from cv_bridge import CvBridge
import numpy as np
import argparse

def save_imu_data(bag, kitti, imu_frame_id, topic):
    print("Exporting IMU")
    for timestamp, oxts in zip(kitti.timestamps, kitti.oxts):
        q = tf.transformations.quaternion_from_euler(oxts.packet.roll, oxts.packet.pitch, oxts.packet.yaw)
        imu = Imu()
        imu.header.frame_id = imu_frame_id
        imu.header.stamp = rospy.Time.from_sec(float(timestamp.strftime("%s.%f")))
        imu.orientation.x = q[0]
        imu.orientation.y = q[1]
        imu.orientation.z = q[2]
        imu.orientation.w = q[3]
        imu.linear_acceleration.x = oxts.packet.af
        imu.linear_acceleration.y = oxts.packet.al
        imu.linear_acceleration.z = oxts.packet.au
        imu.angular_velocity.x = oxts.packet.wf
        imu.angular_velocity.y = oxts.packet.wl
        imu.angular_velocity.z = oxts.packet.wu
        bag.write(topic, imu, t=imu.header.stamp)

def save_imu_data_raw(bag, kitti, imu_frame_id, topic):
    print("Exporting IMU Raw")
    synced_path = kitti.data_path
    unsynced_path = synced_path.replace('sync', 'extract')
    imu_path = os.path.join(unsynced_path, 'oxts')

    # read time stamp (convert to ros seconds format)
    with open(os.path.join(imu_path, 'timestamps.txt')) as f:
        lines = f.readlines()
        imu_datetimes = []
        for line in lines:
            if len(line) == 1:
                continue
            timestamp = datetime.strptime(line[:-4], '%Y-%m-%d %H:%M:%S.%f')
            imu_datetimes.append(float(timestamp.strftime("%s.%f")))

    # fix imu time using a linear model (may not be ideal, ^_^)
    imu_index = np.asarray(range(len(imu_datetimes)), dtype=np.float64)
    z = np.polyfit(imu_index, imu_datetimes, 1)
    imu_datetimes_new = z[0] * imu_index + z[1]
    imu_datetimes = imu_datetimes_new.tolist()

    # get all imu data
    imu_data_dir = os.path.join(imu_path, 'data')
[... 369 lines cut ...]
        if len(kitti.timestamps) == 0:
            print('Dataset is empty? Exiting.')
            sys.exit(1)
            
        if args.sequence in odometry_sequences[:11]:
            print("Odometry dataset sequence {} has ground truth information (poses).".format(args.sequence))
            kitti.load_poses()

        try:
            util = pykitti.utils.read_calib_file(os.path.join(args.dir,'sequences',args.sequence, 'calib.txt'))
            current_epoch = (datetime.utcnow() - datetime(1970, 1, 1)).total_seconds()
            # Export
            if args.kitti_type.find("gray") != -1:
                used_cameras = cameras[:2]
            elif args.kitti_type.find("color") != -1:
                used_cameras = cameras[-2:]

            save_dynamic_tf(bag, kitti, args.kitti_type, initial_time=current_epoch)
            for camera in used_cameras:
                save_camera_data(bag, args.kitti_type, kitti, util, bridge, camera=camera[0], camera_frame_id=camera[1], topic=camera[2], initial_time=current_epoch)

        finally:
            print("## OVERVIEW ##")
            print(bag)
            bag.close()