TixiaoShan/LIO-SAM #342: Starting position error

Cristian-wp, 2022-05-04:
Hello,
I am using LIO-SAM with Mavros on a drone.
I am facing an initial position problem, when my system start I see my initial pose orientation rotated about the World reference frame (identity matrix) about +90° on Z axis.In order to fix this misalignment I need to move and rotate the system on each axis. 
I know that in some cases in order to find the correct Z orientation the IMU must be exited, but I can not understand if this is the case or is pure luck.
Thanks in advance.
[image]

Cristian-wp, 2022-05-09:
I had study your paper and some other things to understand better, and I think that this slam need an inital imu movimentation. Because how GTSAM imupreintegration is implemented, it need some new position to make the factor graph work in the estimation of the initial position.
Is this correct?
@TixiaoShan

cudnn, 2022-07-04:
I am using LIO-SAM with Mavros on a drone too.
i find this code  use  LIO-SAM with Mavros 
https://github.com/lachie-aerialrobotics/tricopter/blob/main/src/Lidar_msg_converter.py
hope help you