TixiaoShan/LIO-SAM #344: On the choice of extrinsicRot & extrinsicRPY in /config/params.yaml

boboboneX, 2022-05-11:
**Greetings to Tixiao for the masterpiece LIO-SAM**! I am having a question regarding the extrinsicRot & extrinsicRPY matrices in /config/params.yaml. Details are below:

# Summary
Should we set both `extrinsicRot` and  `extrinsicRPY` to **identity matrices**? If we are
1. using the recommended IMU ROS pack (`microstrain_mips` of Tixiao-modified version), and
2. aligning the IMU and LiDAR as the photo recommended in REAMDE.md?

# Details

## Coordinate definition in `microstrain_mips`

In issue #128, the recommended ROS pack is mentioned and shared. The pack is derived from `microstrain_mips` (see [https://github.com/ros-drivers/microstrain_mips](url) ) with modifications for LIO-SAM---In particular: the X and Z axes are reversed, which are coded in line 3750 of `microstrain_3dm.cpp`:

```c++
      // tixiao
      // Roboat IMU
      // IMU placement   ----     ROS
      // y ---  z (*)               x
      //        |                   |
      //        |                   |
      //        x             y ----z (x)
      imu_correct_msg_ = imu_msg_;
      imu_correct_msg_.header.frame_id = "base_link";
      // transform angular velocity
      imu_correct_msg_.angular_velocity.x = - imu_msg_.angular_velocity.x;
      imu_correct_msg_.angular_velocity.y =   imu_msg_.angular_velocity.y;
      imu_correct_msg_.angular_velocity.z = - imu_msg_.angular_velocity.z;
      // transform acceleration
      imu_correct_msg_.linear_acceleration.x = - imu_msg_.linear_acceleration.x;
      imu_correct_msg_.linear_acceleration.y =   imu_msg_.linear_acceleration.y;
      imu_correct_msg_.linear_acceleration.z = - imu_msg_.linear_acceleration.z;
      // Obtain orientation
      tf::quaternionMsgToTF(imu_msg_.orientation, orientation);
      beforeMatrix = tf::Matrix3x3(orientation);
      betweenMatrix.setRPY(0, 0, -M_PI/2.0);
      afterMatrix = beforeMatrix * betweenMatrix;
      afterMatrix.getRPY(fixed_roll, fixed_pitch, fixed_yaw);
      geoQuat = tf::createQuaternionMsgFromRollPitchYaw(fixed_roll, fixed_pitch, fixed_yaw);
      imu_correct_msg_.orientation.x = geoQuat.x;
      imu_correct_msg_.orientation.y = geoQuat.y;
      imu_correct_msg_.orientation.z = geoQuat.z;
      imu_correct_msg_.orientation.w = geoQuat.w;
```
## Recommended IMU-LiDAR alignment
In the LIO-SAM README, the recommended IMU-LiDAR alignment is as follows:

[image]

I can see that the hardware alignment follows the reversed X & Z convention (between the IMU and LiDAR frames). Therefore, the modified `microstrain_mips`, playing frame conversion as a ROS node, will gives IMU acc/gyro/quaternion in the LiDAR frame.

## Question
In the LIO-SAM config file (`/config/params.yaml`) , the extrinsicRot and extrinsicRPY are defined, by default, as follows:
```yaml
# Extrinsics: T_lb (lidar -> imu)
  extrinsicTrans: [0.0, 0.0, 0.0]
  extrinsicRot: [-1, 0, 0,
                  0, 1, 0,
                  0, 0, -1]
  extrinsicRPY: [0, -1, 0,
                 1, 0, 0,
                 0, 0, 1]
  # extrinsicRot: [1, 0, 0,
  #                 0, 1, 0,
  #                 0, 0, 1]
  # extrinsicRPY: [1, 0, 0,
  #                 0, 1, 0,
  #                 0, 0, 1]
```

My question is, from the conversion in `microstrain_3dm.cpp` and the recommended IMU-LiDAR alignment, shouldn't we use two identity matrices for extrinsicRot and extrinsicRPY, like the commented lines above? What is the reason for using the current, non-identity matrices?

zyx-99, 2022-05-17:
Hello, I'm also confused by this problem. Have you figured it out now?

boboboneX, 2022-05-17:
Not yet, I will do some tests and report in the results under this issue later.

发件人: zyx-99
发送时间: 2022年5月17日 16:34
收件人: TixiaoShan/LIO-SAM
抄送: boboboneX; Author
主题: Re: [TixiaoShan/LIO-SAM] On the choice of extrinsicRot & extrinsicRPY in /config/params.yaml (Issue #344)

Hello, I'm also confused by this problem. Have you figured it out now?

filesmuggler, 2022-05-27:
hey @boboboneX, have you come up with any results?

boboboneX, 2022-05-29:
Tests are done: the answer is simply that you should keep the default settings, namely:

1. Keep Tixiao's X & Z reverting behavior in the modified `microstrain_mips` pack
2. Follow the recommended IMU-LiDAR alignment
3. Keep the default matrices in `config/params.yaml`, i.e.,
```yaml
extrinsicRot: [-1, 0, 0,
                  0, 1, 0,
                  0, 0, -1]
extrinsicRPY: [0, -1, 0,
                 1, 0, 0,
                 0, 0, 1]
```

So you don't have to do anything extra to make LIO-SAM run correctly. But to understand why reverting twice in both `microstrain_mips` pack and `config/params.yaml`, one should carefully examine the source...

closed 2022-05-29 by boboboneX