TixiaoShan/LIO-SAM #346: imuPreintegration instability

Danielvd598, 2022-05-25:
Hi there, I am a bit confused regarding some reproducability issues with my own dataset.
Incidentally I can complete my dataset without any issues, and the result looks like this:
[image]

However, with the same settings I can also get results like:
[image]
[image]
[image]

With the warning:  Large velocity, reset IMU-preintegration!

Hence,  the instability does not always occur at the same time in the bag (see the pictures above). I checked my extrinsics, which seem to be correct.
Next to that, I use the OS-0 lidar with the PTP protocol, and checked my delay which which is about 0.14 seconds. 
At first I though the issue was similar to #283 , but the fix mentioned by AlexReimann did not resolve the issue. 

Anyways here is the bagfile:
https://drive.google.com/file/d/17tQ8qErCBqSgATalC6Hpg0uOCR4g0oRe/view?usp=sharing

If there is anybody who knows why this could happen, help is very welcome!

Jilei-1996, 2022-08-04:
what compute platform do you use,？i think the problem may be caused by your compute device

joao-aguizo, 2022-11-21:
Was anyone following this topic able to solve this?