TixiaoShan/LIO-SAM #351: High frequency IMU topic acts wierd.

vinci-xiao, 2022-06-23:
Hi, thanks for the great code.
I am using Velodyne VLP-16 lidar and Xsens MTi-3 imu for building map.
When the freq of lidar topic is 10hz and freq of the IMU topic to 10hz, at least the base_link looks fine near the odom.
[image]

But when I increase the freq of the IMU topic from 10 to 100 hz, the base_link goes wrong.*(the path has a spiral pattern even the lidar and imu are standing still)
[image]

Here is my setting:
```
LIDAR
            x
       ^                                     
       |
<------o  z     
y

IMU
            x
       ^                                     
       |
<------o  z     
y
```

The README suggest us to use 500hz on IMU, so I think maybe something wrong with my setting? Or should I just go with 10hz on IMU?

aniketkalra11, 2022-06-23:
I am also facing the same issue even though I am using 200hz of imu data

learnhard22, 2022-09-05:
Hey,

I'm also having this issue with my IMU (SBG ellipseD). The IMU is flickering a little bit in all directions with a frequency of 200Hz. A few changes may need to be done in the settings but I am currently not able to figure out where the problem is exactly. 
My settings of the frame directions are the same as @vinci-xiao.

Did anybody solved this problem already?

All the best,
Lennard

PranavShevkar, 2024-02-23:
I am facing the same issue. Did anyone find any solution