TixiaoShan/LIO-SAM #353: Rotational offset of LIDAR

MrOCW, 2022-07-02:
Hi, I have a Ouster1-32 lidar mounted 45 degrees facing northwest whereas my IMU (MTi630 R) is facing north.
[image]

Will this set up still work with LIO-SAM? If so, are these the right extrinsics?
extrinsicTrans: [x units in lidar's frame to imu, y units in lidar's frame to imu, z units in lidar's frame to imu]
extrinsicRot: [0.707, -0.707, 0,
                  -0.707, 0.707, 0,
                  0, 0, 1]
extrinsicRPY: [1, 0, 0,
                 0, 1, 0,
                 0, 0, 1]
Thanks!

grassjelly, 2022-07-04:
@MrOCW this worked for me https://www.andre-gaschler.com/rotationconverter/

cacao1987, 2022-09-05:
@MrOCW have your solved this prolem?

ArghyaChatterjee, 2022-09-13:
How did you solve your rotational offset issue? It looks more like you have figured out the transformation from lidar to imu. How did you give a tilt transform from base_link to lidar at the first place (As you mentioned your lidar tilted 45 degrees). 

I am also facing the same issue. IF somebody can help, that would be great.

MrOCW, 2022-09-14:
I have not solved it. I just mounted the lidar to face forward in the end. Never had issues with my lidar imu tf

grassjelly, 2022-09-14:
I could’t get decent results with a rotated IMU. I ended up mounting it the same orientation as the LIDAR and use an identity matrix to define the extrinsics.