TixiaoShan/LIO-SAM #361: lidar_link to map transformation is not working

oskarmue, 2022-08-01:
Hey @TixiaoShan thank you for the great work with Lio Sam.
I am using a Robosense Bpearl Lidar with an Xsens IMU. I am using the XYZRIT point type and changed the timestamp to a relative time the same way it is done here: https://github.com/HViktorTsoi/rs_to_velodyne.
The Lidar scans in a frquency of 10Hz, the IMU measures with a frequency of 200Hz. 
When I start Lio Sam, it builds a map, but the map is not accurate at all. 
I think the reason is, that the transformation from lidar link to map is not working right. In rviz it is flickering, so every now and then the transformation is working for a second and then not anymore.
I think my transformation from imu to lidar is correct.
What else should I check? from the Issue page the only thing might be the synchronization part of the imu and lidar, might that be my problem?

[image]

sagarvl96, 2023-06-22:
Hello @oskarmue! I am not sure if you have found a fix for your issue? I also use the data from a Robosense Bpearl Lidar. I didn't understand why you would change to Velodyne pointcloud format. Could you maybe explain a little about this? Thank you!

enginksz, 2024-08-14:
Hello,
I encountered the same problem.
Were you able to solve the problem?