TixiaoShan/LIO-SAM #365: Kitti sequence 00

kefa27, 2022-08-11:
Hi, 

first of all, nice project.
When I test the system on the provided kitti dataset 2011_09_30_drive_0028, all is working fine.
After that I wanted to test the system on the kitti sequence 00 (2011_10_03_drive_0027), which I generated with the provided kitti2bag implementation. I checked the calibration file and it is exactly the same as for the 2011_09_30_drive_0028. However, when I let run the system on kitti sequence 00, the estimated trajectory is jittering a lot and after a while, it is jumping back and forth.
I wonder why this is the case? The whole setup is exactly the same for both datasets.
Do you have any logic conclusion for this? I'm sure you tested your system on sequence 00 as well at some time.
Thanks.

juliangaal, 2022-11-19:
Just related, what's the "drive" to "sequence" logic? How do you know `drive_0027` refers to sequence 0?

fkp0918, 2023-04-14:
i met the same problem, have you solved it?