TixiaoShan/LIO-SAM #366: Livox Mid-70 with 9axis IMU don't work

yuataka, 2022-08-16:
Thank you for publishing this great code, but I have some trouble to run lio-sam with Livox Mid-70 and 9 axis IMU.

First, I checked ```/lio_sam_imageProjection ``` based on this issue.
https://github.com/TixiaoShan/LIO-SAM/issues/110

 ```/lio_sam/deskew/cloud_deskeded```  topic is not outputted from ```/lio_sam_imageProjection ``` .
I also checked input to ```/lio_sam_imageProjection ``` . 
``` /points_raw ``` topic is outputted, but ```/odometry/imu_incremental``` is not outputted.
However, ```/imu``` is outputted correctly and inputted to ```/lio_sam_imuPreintegration```.

Therefore, I think ```/lio_sam_imuPreintegration``` doesn't work correctly because my setting or IMU output have some problem.
[image]

Here is my config and I used your custum livox ros driver.
```
  # Topics
  pointCloudTopic: "points_raw"               # Point cloud data
  imuTopic: "imu"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "livox_frame"
  baselinkFrame: "livox_frame"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useImuHeadingInitialization: false           # if using GPS data, set to "true"
  useGpsElevation: false                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data
  
  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/Downloads/LOAM/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

  # Sensor Settings
  sensor: livox                            # lidar sensor type, 'velodyne' or 'ouster' or 'livox'
  N_SCAN: 1                                  # number of lidar channel (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
  Horizon_SCAN: 3000                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
  downsampleRate: 1                           # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1
  lidarMinRange: 0.3                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 100.0                       # default: 1000.0, maximum lidar range to be used

  # IMU Settings
  imuAccNoise: 3.9939570888238808e-03
  imuGyrNoise: 1.5636343949698187e-03
  imuAccBiasN: 6.4356659353532566e-05
  imuGyrBiasN: 3.5640318696367613e-05
  imuGravity: 9.80511
  imuRPYWeight: 0.01
```

If there is a problem on this config, could you please tell me?

gongyue666, 2022-08-31:
I also try to run with livox mid 70, but it seem do not support lidar data with field "ring".

JasonWang9805, 2022-09-16:
"ring" ：Is it the "line" of livox point cloud data?

525753936, 2022-10-13:
> support lidar data with field "ring".

So did I. Did you try to run fast-lio with livox mid 70 and IMU ? I tried, but it turned our that the odometry would have large drifts once the threads started.

redheli, 2022-10-16:
yes, I tried run with HAP and external imu(3dm-gx5) , large drift, even when the lidar is stationary

qmatias, 2022-11-01:
@yuataka were you able to get LIO-SAM working with the Livox MID-70? Could you share your configuration and a working bag file?

Thanks.

Griwa88, 2023-03-14:
I got Mid-70 to work on outside scan. I think the issue is all the points are quiet dense in a tight beam, this gives too many false positives and causes the drift. Also the rotating pattern isn't much help.

My parameters changes which got it to work are:

  sensor: livox                                # lidar sensor type, 'velodyne' or 'ouster' or 'livox'
  N_SCAN: 1                                  # this is line number in custom msg, there is only 1 value '0' therefore only one 'channel'
  Horizon_SCAN: 9984                  # MID70 9984 points per message
  downsampleRate: 1                    # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1
  lidarMinRange: 0.2                      # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 250.0                 # default: 1000.0, maximum lidar range to be used

  # LOAM feature threshold
  edgeThreshold: 1.0
  surfThreshold: 0.1
  edgeFeatureMinValidNum: 1    # for MID-70
  surfFeatureMinValidNum: 100

  # voxel filter paprams
  odometrySurfLeafSize: 0.2                     # default: 0.4 - outdoor, 0.2 - indoor, 0.2 - outdoor MID-70
  mappingCornerLeafSize: 0.2                  # default: 0.2 - outdoor, 0.1 - indoor, 0.2 - outdoor MID-70
  mappingSurfLeafSize: 0.2                      # default: 0.4 - outdoor, 0.2 - indoor, 0.2 - outdoor MID-70

Also, I wrote a Subscriber/Publisher node to take Livox Custom Message and translate it to PointXYZIRT, but you can use Tixiao custom driver: https://github.com/TixiaoShan/livox_ros_driver [I'm not publishing my Sub/Pub node, maybe in the future once I clean it up, as it was my first go at C++, but the idea is I can use custom message for other localizations like FastLio and LioSam without having to rescan.]
x->x
y->y
z->z
intensity->intensity
line->ring
time / 1000000000->time

Hope this helps.

yuataka, 2023-03-28:
Sorry, currently I am not trying to run lio-sam with Livox Mid-70. So, I close this issue. Thank you.

closed 2023-03-28 by yuataka