TixiaoShan/LIO-SAM #369: A problem of mapping

sushan551, 2022-08-25:
[image]

Hi,
    I am using LIO-SAM, installed on a mobile robot, to build maps of some typical indoor/outdoor environment. One problem that we encountered is described as below:
1. Environment: underground parking lot;
2. IMU: 9-axis Yesense 510-A; Output sampling frequency: 200Hz;
3. LiDAR: RSLidar-16 (Robosense-16)
4. The robot stopped in front of a crossroad, after a relatively long trip, and was supposed to wait for 3 seconds according to my mapping policy. A car run quikly along the perpendicular way suddenly, and the map that the robot was building flied further and further until I shut down the process. Meanwhile, LIO-SAM reported a warning "large velocity detected, reset IMU pre-integration" periodically.
5. I double-checked the source code imuPreintegration.cpp, and found how the warning info was generated. It seemed to me that gtsam had produced a very large velocity. I have no idea for now how to fix this issue.

I am wondering if you could take some time and think about the issue I report here. Any suggestions would be highly appreciated.

Sincerely!

Bag：https://drive.google.com/file/d/1o0oq9sBvExKlUSYxiZ_pciogtBZQyMSL/view?usp=sharing

JaySlamer, 2022-09-15:
@sushan551 Hi
Mind sharing your config file?

legoo-jie, 2022-10-07:
Have you solved the problem? I have a similar problem. Can we talk about it? Thank you very much

cdefg, 2023-05-24:
[image]
like this ?
Map flying in sky.
I think it's a matter of optimazation failure.