TixiaoShan/LIO-SAM #372: units for IMU settings

learnhard22, 2022-09-06:
Hey,

I have a question about the units for IMU settings and I'm not able to assign the values which are given by the IMU manufacturer. I am very new into ROS and SLAM, so please bear with me. Btw I'am using SBG Systems ellipseD.

The following screenshot includes the values that are given by the manufacturer and I think I marked the right ones for the particular parameters. 

[image]

The question for me is now what values were used in the params.yaml file. I think the assignment would be as follows:

- > imuAccNoise is "Velocity Random Walk"
- > imuGyrNoise is "Angular Random Walk"
- > imuAccBiasN is "In run bias Instability" (from acc part above)
- > imuGyrBiasN is "In run bias Instability" (from gyro part below)

Can anybody confirmed my idea or am I totally wrong?

The second idea was to use another package to get the noise values for my IMU, so I used imu_utils. The output is as followed:

[image]

I assumed that gyr_n is for imuGyrNoise and so on (you can see my thoughts for assignment in the screenshot above). 
Are the units of the output of imu_utils are the same as the units of the default values in params.yaml file? 

I am very confused about the units of the IMU manufacturer, so maybe someone can help me here.

Thanks in advantage and kind regards,
Lennard

bilalkah, 2023-01-10:
Hi, Can you find anything to calculate IMU noises?

learnhard22, 2023-01-11:
Hi,
unfortunately not.
I took it like I mentioned above in the last screenshot and I got acceptably values.
So the package I used was imu_utils.

Moraisgit, 2024-04-16:
@learnhard22 Hi! I am having trouble understanding how my output matches yours. I recorded my bag and just played it, maybe the author updated the repo meanwhile and changed how the results are displayed. When I run it I get:

```gyr x  num of Cluster 100
gyr y  num of Cluster 100
gyr z  num of Cluster 100
acc x  num of Cluster 100
acc y  num of Cluster 100
acc z  num of Cluster 100
wait for imu data.
gyr x  numData 478540
gyr x  start_t 1.70483e+09
gyr x  end_t 1.70483e+09
gyr x dt 
-------------6301.32 s
-------------105.022 min
-------------1.75037 h
gyr x  freq 75.9426
gyr x  period 0.0131678
gyr y  numData 478540
gyr y  start_t 1.70483e+09
gyr y  end_t 1.70483e+09
gyr y dt 
-------------6301.32 s
-------------105.022 min
-------------1.75037 h
gyr y  freq 75.9426
gyr y  period 0.0131678
gyr z  numData 478540
gyr z  start_t 1.70483e+09
gyr z  end_t 1.70483e+09
gyr z dt 
-------------6301.32 s
-------------105.022 min
-------------1.75037 h
gyr z  freq 75.9426
gyr z  period 0.0131678
Gyro X 
C  -4.28718   41.4572   14.2237  -1.02915 0.0253362
 Bias Instability 5.38655e-05 rad/s
 Bias Instability 4.13447e-05 rad/s, at 228.172 s
 White Noise 6.92829 rad/s
 White Noise 0.00194964 rad/s
  bias -0.0236426 degree/s
-------------------
Gyro y 
C  -3.94677    38.236    15.608  -1.24558 0.0578972
 Bias Instability 8.99177e-05 rad/s
 Bias Instability 6.7641e-05 rad/s, at 61.6123 s
 White Noise 5.77181 rad/s
 White Noise 0.00168581 rad/s
  bias 0.238684 degree/s
-------------------
Gyro z 
C  -3.82363   37.0779   11.2089 -0.936989 0.0235316
 Bias Instability 2.89027e-05 rad/s
 Bias Instability 2.63716e-05 rad/s, at 202.574 s
 White Noise 6.53143 rad/s
 White Noise 0.00182199 rad/s
  bias 0.0479615 degree/s
-------------------
==============================================
==============================================
acc x  numData 478540
acc x  start_t 1.70483e+09
acc x  end_t 1.70483e+09
acc x dt 
-------------6301.32 s
-------------105.022 min
-------------1.75037 h
acc x  freq 75.9426
acc x  period 0.0131678
acc y  numData 478540
acc y  start_t 1.70483e+09
acc y  end_t 1.70483e+09
acc y dt 
-------------6301.32 s
-------------105.022 min
-------------1.75037 h
acc y  freq 75.9426
acc y  period 0.0131678
acc z  numData 478540
acc z  start_t 1.70483e+09
acc z  end_t 1.70483e+09
acc z dt 
-------------6301.32 s
-------------105.022 min
-------------1.75037 h
acc z  freq 75.9426
acc z  period 0.0131678
acc X 
C -0.000787586   0.00488469    -0.000186  4.49406e-05  2.68223e-06
 Bias Instability 0.000932914 m/s^2
 White Noise 0.0340983 m/s^2
-------------------
acc y 
C -0.000396469   0.00264456  3.95637e-05  3.82576e-05 -1.81843e-07
 Bias Instability 0.000771288 m/s^2
 White Noise 0.0228882 m/s^2
-------------------
acc z 
C -0.000453912   0.00284743 -0.000129331  5.10706e-05  7.65481e-07
 Bias Instability 0.000670796 m/s^2
 White Noise 0.019311 m/s^2
-------------------
[imu_an-1] process has finished cleanly
log file: /home/morais/.ros/log/ad749cdc-fc0e-11ee-a90d-191d9557722a/imu_an-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
I think it is only missing the average values that you have on your output, so to get the average noise and bias, it would only be necessary to do a average of the values of the 3 axis right?

Thank you in advance!