TixiaoShan/LIO-SAM #375: Confused about the "increOdomAffine" in mapOptmization.cpp

Lurvelly, 2022-09-09:
Hi, Tixiao Shan:

Thanks for sharing your code. It's very useful.
During learning your paper and code, I'm wonder the meaning of the code below.

https://github.com/TixiaoShan/LIO-SAM/blob/9991daa47eb76124659570daab299bf877dde8bb/src/mapOptmization.cpp#L1663-L1664

In my view, the "increOdomAffine" and the "incrementalOdometryAffineFront" are the same before program go through the line 1663 and 1664.

At the first time code reach the line 1663, the "increOdomAffine" is equal to the previous frame's "transformTobeMapped", at the same time, the "incrementalOdometryAffineFront" is on record as previous frame's "transformTobeMapped".
https://github.com/TixiaoShan/LIO-SAM/blob/9991daa47eb76124659570daab299bf877dde8bb/src/mapOptmization.cpp#L788-L789

And also, we can simplify the code at line 1663 - 1664 like:
increOdomAffine = increOdomAffine * incrementalOdometryAffineFront.inverse() * incrementalOdometryAffineBack;

According to previous analysis, the "increOdomAffine" is equal to the "incrementalOdometryAffineFront", so we can write as: 
increOdomAffine = incrementalOdometryAffineFront * incrementalOdometryAffineFront.inverse() * incrementalOdometryAffineBack;

As all known the "incrementalOdometryAffineFront * incrementalOdometryAffineFront.inverse()" is identity. So the 
"increOdomAffine" is equal to the "incrementalOdometryAffineBack" which is equal to the "transformTobeMapped" that computed at the scan2MapOptimization function.

So, I think we can just simplify the code at line 1663 - 1664 to:  
    increOdomAffine = trans2Affine3f(transformTobeMapped); 

Previous code will bring some error of calculation and this will be more clear and more accurate, isn't it?

Thanks in advantage and kind regards,
Liuyang Li

JaySlamer, 2022-09-15:
@Lurvelly 
`increOdomAffine = trans2Affine3f(transformTobeMapped)`
**is totally different from**
```
Eigen::Affine3f affineIncre = incrementalOdometryAffineFront.inverse() * incrementalOdometryAffineBack; 
increOdomAffine = increOdomAffine * affineIncre;
```
Estimation of biases of imu need lidar odometry to help constraint it. However, pose graph optimization using GPS would cause transformTobeMapped to jump hugely and suddenly, which will cause failure of estimation of biases of imu. So it needs to use a version of lidar odometry without being affected by pose graph optimization, that is the laserOdomIncremental.

Lurvelly, 2022-09-15:
@JaySlamer
Dear Jay,

Thanks for your reply.

Yes, you are right. 
That's my mistake, I haven't noticed that pose graph optimization would change the transformTobeMapped, so I was simple enough to think this question.

I will close this issue, thanks again.

> @Lurvelly `increOdomAffine = trans2Affine3f(transformTobeMapped)` **is totally different from**
>
> ```
> Eigen::Affine3f affineIncre = incrementalOdometryAffineFront.inverse() * [...]

closed 2022-11-08 by Lurvelly