TixiaoShan/LIO-SAM #378: Using Ouster's internal IMU

Dhruv0208, 2022-09-28:
Hi, I am looking for make the use of Ouster LIDAR's internal IMU. But as mentioned, using VINS-Mono can help in running LIO-SAM with 6-axis IMU. Can you explain about this a bit more, about how it can be done?

Zermelo-101, 2023-03-20:
Did you find a solution ?

CantLooseToAMoose, 2024-06-03:
> > Did you find a solution ?
> 
> Hi, did you find a solution?

Hey, did you find a solution?

ZhangYaoll, 2024-06-04:
there is projection helped me solve the problem: https://github.com/YJZLuckyBoy/liorf.

CantLooseToAMoose, 2024-06-04:
> there is projection helped me solve the problem: https://github.com/YJZLuckyBoy/liorf.

Hey I tried out liorf, but when trying to construct the Pointcloud a lot of jumping is happening. See [#41](https://github.com/YJZLuckyBoy/liorf/issues/41). Can you maybe share your config file?