TixiaoShan/LIO-SAM #380: Ros Noetic

Abhirup0209, 2022-10-04:
Hello @TixiaoShan ,

Can you please help me as I have followed all the instructions as per directed to compile the package on ROS Noetic but somehow, the system hangs and the compilation does not proceed.

Can anyone please help me to solve this issue, as it ran perfectly in July but now it is not compiling properly. I have chosen this algorithm as a part of my Master Thesis, so it is really important.

bestavra, 2022-10-09:
Hi @Abhirup0209, what error did you get? I've successfully compiled on Noetic with catkin build? I had to adjust some things though:
1. In include/utility.h change opencv header to `#include <opencv2/opencv.hpp`
2. In the same file move `#include <pcl/kdtree/kdtree_flann.h>` right before opencv header line
3. In CMakeLists.txt change c++ version to 14

Hope this helps.

Abhirup0209, 2022-10-09:
https://blog.csdn.net/zardforever123/article/details/125570551?spm=1001.2101.3001.6650.3&utm_medium=distribute.pc_relevant.none-task-blog-2%7Edefault%7ECTRLIST%7ERate-3-125570551-blog-116279537.t0_edu_mix&depth_1-utm_source=distribute.pc_relevant.none-task-blog-2%7Edefault%7ECTRLIST%7ERate-3-125570551-blog-116279537.t0_edu_mix&utm_relevant_index=4

Thanks, but now I have successfully compiled on Ros Noetic with the settings and changes mentioned in the blogpost attached to this comment. Whoever else working on Noetic, can kindly look into the issue.

closed 2022-10-09 by Abhirup0209